zlgopen/awtk #10: imagegen tool 不支持本身是RGB565 bmp格式的转变为 .data文件

zhanglu993, 2018-09-22:
imagegen tool 不支持RGB565 bmp格式图片的转变为 .data文件，提示image_loader_stb_load:84 data != NULL 的错误。
如果直接使用 imagegen tool 转变 RGBA8888格式的图片为 RGB565格式的  .data文件，是可以正确产生文件，但是会出现颜色严重失真的问题。
不能转变的RGB565 bmp文件 下载地址 链接：https://pan.baidu.com/s/1u_rXB80tyecGxGG1gZCiLQ 密码：e8ua

xianjimli, 2018-09-22:
谢谢，今天我在移植AWTK到STM32F767上，明天上午我看看。

zhanglu993, 2018-09-22:
好的，辛苦了，不知道为什么通过imagegen tool转换后颜色失真会这么严重，在电脑上用Image2Lcd 转换565的效果还行。

xianjimli, 2018-09-23:
这是个STB图片库不支持的压缩的BMP文件，我转成png或jpg格式没有问题，你转成png或jpg吧。
 
thread #1: tid = 0xa4c78, 0x000000010000184a image_info`stbi__err(str="bad BMP") + 10 at stb_image.h:844, queue = 'com.apple.main-thread', stop reason = breakpoint 3.1
  * frame #0: 0x000000010000184a image_info`stbi__err(str="bad BMP") + 10 at stb_image.h:844
    frame #1: 0x000000010001680f image_info`stbi__bmp_parse_header(s=0x00007fff5fbff7e0, info=0x00007fff5fbff1e0) + 943 at stb_image.h:5049
    frame #2: 0x0000000100002515 image_info`stbi__bmp_load(s=0x00007fff5fbff7e0, x=0x00007fff5fbff950, y=0x00007fff5fbff94c, comp=0x00007fff5fbff954, req_comp=0, ri=0x00007fff5fbff758) + 149 at stb_image.h:5088
    frame #3: 0x0000000100001e24 image_info`stbi__load_main(s=0x00007fff5fbff7e0, x=0x00007fff5fbff950, y=0x00007fff5fbff94c, comp=0x00007fff5fbff954, req_comp=0, ri=0x00007fff5fbff758, bpc=8) + 276 at stb_image.h:973
    frame #4: 0x00000001000019e8 image_info`stbi__load_and_postprocess_8bit(s=0x00007fff5fbff7e0, x=0x00007fff5fbff950, y=0x00007fff5fbff94c, comp=0x00007fff5fbff954, req_comp=0) + 72 at stb_image.h:1063
    frame #5: 0x00000001000018da image_info`stbi_load_from_file(f=0x00007fff7c035050, x=0x00007fff5fbff950, y=0x00007fff5fbff94c, comp=0x00007fff5fbff954, req_comp=0) + 122 at stb_image.h:1149