zlgopen/awtk #13: stm32f103ze之类的低配置平台是否支持demoui的样式工程？

zhanglu993, 2018-10-16:
你好，想请问stm32f103ze之类的低配置平台是否支持demoui的样式工程？我现在使用开发板放入asset.c和demo_ui_app.c来测试，程序可以很顺利的进入主界面，以及各个子界面，但是当点击close退出 Basic，Edits等子窗口时，主界面不会刷回正常界面，只会显示点击进入界面的按钮
图1 启动进入主界面
[image]

图2 进入basic control界面
[image]

图3 按close退出后主界面显示，显示有问题
[image]

xianjimli, 2018-10-17:
可用内存有多少，如果只有内置的32KRAM，估计不行。

zhanglu993, 2018-10-17:
现在是分配了32K，Basic控件本身消耗的内存会超过这个32K么？

xianjimli, 2018-10-17:
控件本身肯定够的。全给了GUI，stack会不会重叠了。没这样玩过，你先看看。

zhanglu993, 2018-10-17:
我片子上总共有 65K内存，给了GUI 36K 其他地方应该还是够的。

zhanglu993, 2018-10-22:
我debug了一下代码，发现可能是初始化的问题，但不知道这样修改是否合适，还请看一下

原始代码是
ret_t widget_paint(widget_t* widget, canvas_t* c) {
  uint8_t save_alpha = 0;
  return_value_if_fail(widget != NULL && c != NULL, RET_BAD_PARAMS);

  if (!widget->visible || widget->w <= 0 || widget->h <= 0) {
    widget->dirty = FALSE;
    return RET_OK;
  }

  save_alpha = c->lcd->global_alpha;

  if (widget->opacity < TK_OPACITY_ALPHA) {
    uint8_t alpha = (widget->opacity * save_alpha) / 0xff;
    canvas_set_global_alpha(c, alpha);
  }

后来增加一个赋值即可正常显示
ret_t widget_paint(widget_t* widget, canvas_t* c) {
  uint8_t save_alpha = 0;
  return_value_if_fail(widget != NULL && c != NULL, RET_BAD_PARAMS);

  if (!widget->visible || widget->w <= 0 || widget->h <= 0) {
    widget->dirty = FALSE;
    return RET_OK;
  }
  else   //增加一个赋值条件
  {
      widget->dirty = TRUE;
   }

  save_alpha = c->lcd->global_alpha;

  if (widget->opacity < TK_OPACITY_ALPHA) {
    uint8_t alpha = (widget->opacity * save_alpha) / 0xff;
    canvas_set_global_alpha(c, alpha);
  }

xianjimli, 2018-10-23:
谢谢。可能是用USE_FAST_MODE启用极速模式引起的，极速模式可能存在一些潜在的问题。你#undef USE_FAST_MODE试试。

zhanglu993, 2018-10-23:
我试了，看了下debug的值，在我的环境中  widget->dirty 如果没有走为 FALSE这个条件，widget->dirty对应的值也有可能是FALSE，从而使的整个流程出现问题。

xianjimli, 2018-10-23:
不定义USE_FAST_MODE时，渲染时会自动忽略widget->dirty标志。