zlgopen/awtk #14: imagegen 编译资源报错

zhanglu993, 2018-10-18:
更新到最新版本imagegen后发现编译报错，具体如下：
(base) F:\sourcecode\awtk\original\awtk_new>F:\sourcecode\awtk\original\awtk_new\bin\imagegen.exe F:\sourcecode\awtk\original\awtk_new\demos\assets\raw\images\x1\anib.png F:\sourcecode\awtk\original\awtk_new\demos\assets\inc\images\anib.data bgra|bgr565
'bgr565' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

将bgra|bgr565改成bgra+bgr565后可以正常编译
(base) F:\sourcecode\awtk\original\awtk_new>F:\sourcecode\awtk\original\awtk_new\bin\imagegen.exe F:\sourcecode\awtk\original\awtk_new\demos\assets\raw\images\x1\anib.png F:\sourcecode\awtk\original\awtk_new\demos\assets\inc\images\anib.data bgra+bgr565
done

xianjimli, 2018-10-18:
已修改，谢谢:)

zhanglu993, 2018-10-22:
问题已解决

closed 2018-10-22 by zhanglu993