zlgopen/awtk #17: 最新代码配合awtk-linux-fb启动奔溃

iamhw, 2018-10-26:
awtk版本：
commit 4dc0d227802684c305a22770eaf84579d35bebe6
Date:   Fri Oct 26 07:38:08 2018 +0800

awtk-linux-fb版本：
commit d341737f678987b6f000f1bb7fadaaa9b93ed364
Date:   Wed Oct 24 17:58:43 2018 +0800

启动打印：
/mnt/project/awtk-linux-fb/release/bin # ./demoui 
fb_info_t: /dev/fb0
fb_info_t: xres=1280 yres=720 bits_per_pixel=32 mem_size=1843200
fb_info_t: red(16 8) green(8 8) blue(0 8)
xres_virtual =1280 yres_virtual=720 xpanstep=1 ywrapstep=0
Build at: Oct 26 2018 10:03:10
window preload open
Segmentation fault (core dumped)

栈回溯信息
Core was generated by `./demoui'.
Program terminated with signal 11, Segmentation fault.
#0  0x000100f8 in tk_memcpy32 (dst=0xb6a02000, src=<optimized out>, size=228352) at awtk/src/base/utils.c:365
365	    *d++ = *s++;
(gdb) bt
#0  0x000100f8 in tk_memcpy32 (dst=0xb6a02000, src=<optimized out>, size=228352) at awtk/src/base/utils.c:365
#1  0x00010158 in tk_pixel_copy (dst=dst@entry=0xb6a02000, src=src@entry=0xb683f008, size=<optimized out>, bpp=<optimized out>) at awtk/src/base/utils.c:380
#2  0x0005371c in soft_copy_image (dst=0xbef6b8f0, src=0xbef6b920, src_r=0x20c720, dx=<optimized out>, dy=0) at awtk/src/blend/soft_g2d.c:67
#3  0x00051a50 in lcd_mem_flush (lcd=0x20c678) at awtk/src/lcd/lcd_mem.inc:305
#4  0x0004fdf0 in lcd_linux_fb_flush (lcd=<optimized out>) at awtk-port/lcd_linux_fb.c:149
#5  0x0002fc88 in lcd_end_frame (lcd=<optimized out>) at awtk/src/base/lcd.c:194
#6  0x0002a0bc in canvas_end_frame (c=c@entry=0x1aac24 <s_main_loop_simple.9557+36>) at awtk/src/base/canvas.c:919
#7  0x00015e34 in window_manager_paint_normal (c=0x1aac24 <s_main_loop_simple.9557+36>, widget=0x20c528) at awtk/src/base/window_manager.c:305
#8  window_manager_paint (widget=0x20c528, c=c@entry=0x1aac24 <s_main_loop_simple.9557+36>) at awtk/src/base/window_manager.c:394
#9  0x00050668 in main_loop_simple_run (l=0x1aac00 <s_main_loop_simple.9557>) at awtk/src/main_loop/main_loop_simple.c:167
#10 0x00023abc in tk_run () at awtk/src/awtk.c:153
#11 0x00009928 in main () at awtk/demos/demo_main.c:72

xianjimli, 2018-10-26:
awtk-linux-fb是最新版本吗？

iamhw, 2018-10-27:
> 
> 
> awtk-linux-fb是最新版本吗？

是的

awtk-linux-fb版本：
commit d341737f678987b6f000f1bb7fadaaa9b93ed364
Date: Wed Oct 24 17:58:43 2018 +0800

xianjimli, 2018-10-28:
我没能重现，从下面的log信息，猜测是(fb)->fix.line_length有问题：
```
fb_info_t: xres=1280 yres=720 bits_per_pixel=32 mem_size=1843200
```

```
#define fb_size(fb) ((fb)->var.yres * (fb)->fix.line_length)
mem_size = fb_size(fb);
```
mem_size至少是：1280 *720* 4。

我加了点log，你有空试试。

iamhw, 2018-10-30:
嗯，已解决
在fb_open中fb->var.bits_per_pixel从默认的16修改为32后
FBIOPUT_VSCREENINFO后需要重新FBIOGET_FSCREENINFO，因为fb->fix.line_length已经改变

closed 2018-10-30 by iamhw