zlgopen/awtk #26: nanovg_agge对rgb565格式不支持导致界面显示异常，stm32f429 发生hardfault的错误

YasirLiang, 2018-10-31:
你好!调试发现nanovg_agge对rgb565格式不支持，在函数执行断言后就出现了段错误！不过在pc上模拟是通过的。烧写到stm32f429igt6的调试过程如下面的图：

[image]
[image]
    Moltanisk 2018/10/30

xianjimli, 2018-10-31:
麻烦确认一下，LCD是否用lcd_mem_bgr565_create_three_fb创建的。

请参考: https://github.com/zlgopen/awtk-stm32f429igtx-raw/blob/master/USER/main.c

YasirLiang, 2018-11-01:
你好！不是，是用lcd_mem_rgb565_create_three_fb创建的，用的是例程里默认的lcd_mem_rgb565_create_three_fb不是lcd_mem_bgr565_create_three_fb

xianjimli, 2018-11-01:
好的。后来为了统一命名方式(按各颜色通道的内存地址从低到高)，改了函数名。现在AWTK的接口基本稳定了，以后这类不兼容的改动比较少了。

closed 2018-11-01 by xianjimli