zlgopen/awtk #57: 使用bsvggen工具转化svg图片，在程序中使用转化后的图片报错：

liusha0002000, 2018-11-30:


liusha0002000, 2018-11-30:
bsvggen没提示信息，文件怎么发给你？

xianjimli, 2018-12-01:
把内容贴到这里吧

liusha0002000, 2018-12-01:
用bsvgdump显示我的bsvg文件得到信息如下：
bsvg_init:27 svg->header->magic == BSVG_MAGIC
bsvg_to_svg:296 bsvg_init(&svg, data, size) != NULL
<svg viewPoint="544366952 1734438249 1701273445 542989166" width="175841341" height="842037296">
</svg>
这个是表示执行图片转化的时候就出错了吧？

xianjimli, 2018-12-01:
是的，出错了。贴下原始svg文件的内容。

liusha0002000, 2018-12-01:
确实没加第三个参数bin，不过加了生成的也不对，原始文件如下：
`<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg version="1.1" id="Layer_1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0px" y="0px" width="240px" height="240px" viewBox="0 0 240 240" enable-background="new 0 0 240 240" xml:space="preserve">  <image id="image0" width="240" height="240" x="0" y="0"
    xlink:href="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAPAAAADwCAIAAACxN37FAAAABGdBTUEAALGPC/xhBQAAACBjSFJN
AAB6JgAAgIQAAPoAAACA6AAAdTAAAOpgAAA6mAAAF3CculE8AAAABmJLR0QA/wD/AP+gvaeTAACA
...
...
OjQ0KzA4OjAwX1fv+QAAACV0RVh0ZGF0ZTptb2RpZnkAMjAxOC0xMS0zMFQxOToxODo0NCswODow
MC4KV0UAAAAASUVORK5CYII=" />
</svg>`
文件太大，把其中内容省略了大部分

xianjimli, 2018-12-02:
这个不支持。它内嵌的是一张png图片。data:image/png;base64,

liusha0002000, 2018-12-02:
额，我的svg是用png图片转过去的，那支持的svg要怎么制作呢？

xianjimli, 2018-12-03:
PNG转成的SVG都是伪SVG。为什么要把PNG转换成SVG呢？

我是在：http://www.iconfont.cn/ 下载的SVG图片。

liusha0002000, 2018-12-03:
了解了，从你发的链接下载的svg是可以正常显示的，不过好像只能显示32*32px的图片，好在可以放大，这点也没啥影响，谢谢了

liusha0002000, 2018-12-04:
svg图片可以显示大于32*32px的，资源文件修改过以后，最好先执行./scripts/update_res.py clean然后再./scripts/update_res.py all生成资源

closed 2018-12-04 by liusha0002000