zlgopen/awtk #62: combo_box 控件的有两个bug

YasirLiang, 2018-12-17:
combo_box 控件的有两个bug，其一，下拉选择弹出菜单后，滚动combox选项，就直接选中了当前单击的combox item，导致无法选择combox的其他选项；其二，选中combox item后tr_text的内容没有更新，还是上一次选中的内容

xianjimli, 2018-12-18:
第一个问题，我没重现，不知是不是没明白你的重现方法。

第二个问题，我这样写是可以的(如果你写法不同，麻烦贴上XML我看看)
```
<combo_box readonly="true" x="10" y="bottom:5" w="200" h="30" tr_text="ok" options="1:ok;2:cancel;"/>
```
谢谢

YasirLiang, 2018-12-18:
XML代码：
`<combo_box name="edit:selectChannel" style="unitChineseComboBox" open_window="discuss_select_chn_popup" readonly="true" x="45%" y="44%" w="88" h="36" tr_text="全部">
		</combo_box>`

xianjimli, 2018-12-19:
tr_text是翻译前的key，在strings/strings.xml需要定义，并且有相应的字体。你参照我这个试试：
```
<combo_box open_window="language" readonly="true" x="10" y="bottom:50" w="200" h="30" tr_text="english"/>
```

弹出窗口文件：
> demos/assets/raw/ui/language.xml
```
<popup close_when_click_outside="true" h="80" >
  <list_view x="0"  y="0" w="100%" h="100%" item_height="30">
    <scroll_view name="view" x="0"  y="0" w="-12" h="100%">
      <combo_box_item tr_text="english"/>
      <combo_box_item tr_text="chinese" />
    </scroll_view>
    <scroll_bar_d name="bar" x="right" y="0" w="12" h="100%" value="0"/>
  </list_view>
</popup>
```

xianjimli, 2018-12-19:
你看下demoui里的combox在你那里正常吗？

YasirLiang, 2018-12-21:
注册例程combox的“EVT_VALUE_CHANGED”事件，选中“green”log_debug出来的是“red”，在选中“blue”log_debug出来的“green”。这是

xianjimli, 2018-12-22:
已修改。谢谢

YasirLiang, 2018-12-22:
设置combo 的options是可以了，但是不支持open_window的combo的item改变，需要改变combo `的text后再调用`combo_box_set_selected_index`才OK，估计`combo_box_set_selected_index`外部函数不太完善吧。

xianjimli, 2018-12-22:
不会啊。demoui中下面两个combobox就是用open_window打开的。

YasirLiang, 2019-01-03:
不好意思，我更新了，用open_window打开的combobox获取内容还是不正常

YasirLiang, 2019-01-04:
>我

YasirLiang, 2019-01-04:
_我_

YasirLiang, 2019-01-04:
> 修改combo_box_on_item_click函数如下：

```c
static ret_t combo_box_on_item_click(void* ctx, event_t* e) {
  combo_box_t* combo_box = COMBO_BOX(ctx);
  widget_t* widget = WIDGET(combo_box);
  widget_t* item = WIDGET(e->target);
  int32_t index = widget_index_of(item);

  if (item->tr_text != NULL) {
    widget_set_tr_text(widget, item->tr_text);
  } else {
    widget_set_text(widget, item->text.str);
  }

  combo_box->value = COMBO_BOX_ITEM(item)->value;
  combo_box_set_selected_index(WIDGET(combo_box), index);

  window_close(widget_get_window(item));

  return RET_OK;
}

```

就可以了

xianjimli, 2019-01-04:
明白了。已修改。谢谢。

YasirLiang, 2019-01-16:
可以了

closed 2019-01-16 by xianjimli

chensheng-GitHub-2019, 2019-12-25:
combox应该少了一个api接口，例comboBox->currentIndex();参考自qt

xianjimli, 2019-12-25:
> currentIndex

可以直接读取 selected_index 属性。