zlgopen/awtk #64: 往scroll_view增加多个radio_button后无法正常显示radio_button

YasirLiang, 2018-12-17:
往scroll_view增加3个radio_button，显示正常。增加多于3个radio_button时，radio_button之间的间隔不增大，若继续增加到12时，所有的radio_button都不显显示，也无法滚动。根据现象推断scroll_view被设置成水平方向而不是垂直方向，更新到2742784878ad5cd008091f04609b0cac935f081a就变成这样了。

xianjimli, 2018-12-18:
是ListView吗，我放在ListView里没问题。麻烦把XML贴上来看看。

YasirLiang, 2018-12-18:
是listview，代码是这样的：
<list_view style="" x="c"  y="18%" w="92%" h="77%">
	<scroll_view style="" x="c"  y="m" w="100%" h="100%" layout="c:1">
		<radio_button name="radioBtn:1200bps" x="c" y="0%" w="70%" h="33%" style="RightEnglishStyle" text="1200 bps"/>
		<radio_button name="radioBtn:2400bps" x="c" y="33%" w="70%" h="33%" style="RightEnglishStyle" text="2400 bps"/>
		<radio_button name="radioBtn:4800bps" x="c" y="66%" w="70%" h="33%" style="RightEnglishStyle" text="4800 bps"/>
		<radio_button name="radioBtn:9600bps" x="c" y="99%" w="70%" h="33%" style="RightEnglishStyle" text="9600 bps"/>
		<radio_button name="radioBtn:14400bps" x="c" y="133%" w="70%" h="33%" style="RightEnglishStyle" text="14400 bps"/>
		<radio_button name="radioBtn:19200bps" x="c" y="166%" w="70%" h="33%" style="RightEnglishStyle" text="19200 bps"/>
		<radio_button name="radioBtn:38400bps" x="c" y="199%" w="70%" h="33%" style="RightEnglishStyle" text="38400 bps"/>
		<radio_button name="radioBtn:43000bps" x="c" y="233%" w="70%" h="33%" style="RightEnglishStyle" text="43000 bps"/>
		<radio_button name="radioBtn:57600bps" x="c" y="266%" w="70%" h="33%" style="RightEnglishStyle" text="57600 bps"/>
		<radio_button name="radioBtn:76800bps" x="c" y="299%" w="70%" h="33%" style="RightEnglishStyle" text="76800 bps"/>
		<radio_button name="radioBtn:115200bps" x="c" y="333%" w="70%" h="33%" style="RightEnglishStyle" text="115200 bps"/>
		<radio_button name="radioBtn:128000bps" x="c" y="366%" w="70%" h="33%" style="RightEnglishStyle" text="128000 bps"/>
	</scroll_view>
	<scroll_bar_m name="bar" x="right" y="0%" w="4" h="80%" value="0"/>
</list_view>

xianjimli, 2018-12-19:
radio_button不需要指定布局参数，这样就行了：
```
<window anim_hint="htranslate">
  <list_view x="0"  y="0" w="100%" h="-50" item_height="30">
    <scroll_view name="view" x="0"  y="0" w="-12" h="100%">
    <radio_button name="radioBtn:1200bps" style="RightEnglishStyle" text="1200 bps"/>
    <radio_button name="radioBtn:2400bps" style="RightEnglishStyle" text="2400 bps"/>
    <radio_button name="radioBtn:4800bps" style="RightEnglishStyle" text="4800 bps"/>
    <radio_button name="radioBtn:9600bps" style="RightEnglishStyle" text="9600 bps"/>
    <radio_button name="radioBtn:14400bps" style="RightEnglishStyle" text="14400 bps"/>
    <radio_button name="radioBtn:19200bps" style="RightEnglishStyle" text="19200 bps"/>
    <radio_button name="radioBtn:38400bps" style="RightEnglishStyle" text="38400 bps"/>
    <radio_button name="radioBtn:43000bps" style="RightEnglishStyle" text="43000 bps"/>
    <radio_button name="radioBtn:57600bps" style="RightEnglishStyle" text="57600 bps"/>
    <radio_button name="radioBtn:76800bps" style="RightEnglishStyle" text="76800 bps"/>
    <radio_button name="radioBtn:115200bps" style="RightEnglishStyle" text="115200 bps"/>
    <radio_button name="radioBtn:128000bps" style="RightEnglishStyle" text="128000 bps"/>
    </scroll_view>
    <scroll_bar_d name="bar" x="right" y="0" w="12" h="100%" value="0"/>
  </list_view>
</window>
```

> 如果你想控制radio\_button的精确布局，可以把radio\_button放到ListItem中。请参考： demos/assets/raw/ui/list_view_d.xml

YasirLiang, 2018-12-19:
通过去掉radio button本身的布局参数后正常显示了，谢谢🙏。

closed 2018-12-20 by xianjimli