zlgopen/awtk #78: 如何加载不同大小的字体

YasirLiang, 2019-01-16:
由于资源的限制和根据字体显示的需要，需要单独加载所需的不同大小的字体而不是加载ttf文件中的所有字体，请问需要怎么做呢？

xianjimli, 2019-01-16:
对于位图字体，我的想法是用多个字体文件来表示不同大小的字体。比如

* 缺省字体 default.data
* 小字体 small.data
* 大字体 big.data

YasirLiang, 2019-01-17:
那如何编译和应用这些字体资源

xianjimli, 2019-01-17:
* 1.通过fontgen生成数据。
* 2.在demos/assets.c中引入。
* 3.在style中引用。

可以修改一下脚本scripts/update_res_common.py。

目前demoui的大小已经逼近1M了，再大就放不进stm32429/767了，所以没法内置到里面。

YasirLiang, 2019-01-18:
实际使用时使用芯片内部的空间可能不够用，需要使用外部的flash，因此需要把资源文件加载到外部flash中。按照你的三个步骤：
1、在脚本中增加生成相应资源文件的代码
    fontgen('fonts/simhei.ttf', 'fonts/simhei23.txt', 'fonts/simhei23.data', 23);
2、在assets.c包含和引用资源文件
    #include "assets/inc/fonts/simhei23.data"；
    assets_manager_add(rm, font_simhei23);
3、  在style中引用
<style name="BiggestBtnLabelStype">
    <normal text_color="#B9451C" font_name="simhei23" text_align_h="center"/>
 </style>
4、禁止库和应用程序使用文件系统，模拟器运行后字体无法显示，其他资源加载正常。debug串口出现
vgcanvas_nanovg_set_font:257 font_id >=0的信息。

请问是哪个步骤出问题了还是要进行一些其他的操作。

YasirLiang, 2019-01-18:
实际使用时使用芯片内部的空间可能不够用，需要使用外部的flash，因此需要把资源文件加载到外部flash中。按照你的三个步骤：

    1、在脚本中增加生成相应资源文件的代码
fontgen('fonts/simhei.ttf', 'fonts/simhei23.txt', 'fonts/simhei23.data', 23);

    2、在assets.c包含和引用资源文件
#include "assets/inc/fonts/simhei23.data"；
assets_manager_add(rm, font_simhei23);

    3、 在style中这样引用
<normal text_color="white" font_name="simhei23" text_align_h="center"/>

    4、禁止库和应用程序使用文件系统，模拟器运行后字体无法显示，其他资源加载正常。debug串口出现"vgcanvas_nanovg_set_font:257 font_id >=0"的信息。

请问是哪个步骤出问题了还是要进行一些其他的操作？

YasirLiang, 2019-01-18:
实际使用时使用芯片内部的空间可能不够用，需要使用外部的flash，因此需要把资源文件加载到外部flash中。按照你的三个步骤：

    1、在脚本中增加生成相应资源文件的代码：fontgen('fonts/simhei.ttf', 'fonts/simhei23.txt', 'fonts/simhei23.data', 23);

    2、在assets.c包含和引用资源文件：#include "assets/inc/fonts/simhei23.data"；assets_manager_add(rm, font_simhei23);

    3、 在style中这样引用<normal text_color="white" font_name="simhei23" text_align_h="center"/>

    4、禁止库和应用程序使用文件系统，模拟器运行后字体无法显示，其他资源加载正常。debug串口出现"vgcanvas_nanovg_set_font:257 font_id >=0"的信息。

请问是哪个步骤出问题了还是要进行一些其他的操作？

xianjimli, 2019-01-19:
我也不知道如何使用外部的flash，要不就引入文件系统来管理外部flash。

YasirLiang, 2019-01-21:
使用文件系统很慢。使用的外部flash的话，链接所有资源到外部flash的代码段中，使用section编译属性即可，然后在初始化中首先加载资源到之前的链接地址上，这个地址可以是外部内存的地址。

这里的问题是如何能正常加载之前生成的字体资源，在使用模拟器字体也能正常显示。ps：加载.res字体资源文件可以正常显示。