zlgopen/awtk #81: svg demo behavior diff between win and ubuntu(SVG在WIN上demo异常)

jasonbu, 2019-01-22:
start with windows
[image]
process with windows
[image]
process with ubuntu
[image]
SVG demo run on win, lost some details.
SVG demo在windows运行的时候细节丢失.

xianjimli, 2019-01-22:
请参考FAQ第五条：https://github.com/zlgopen/awtk/blob/master/docs/faq.md

jasonbu, 2019-01-22:
```
#NANOVG_BACKEND='GL3'
#NANOVG_BACKEND='GLES2'
#NANOVG_BACKEND='GLES3'
NANOVG_BACKEND='AGG'
#NANOVG_BACKEND='AGGE'
#NANOVG_BACKEND='BGFX'
```
更新SConstruct之后问题解决

closed 2019-01-22 by jasonbu