zlgopen/awtk #89: 如何支持显示摄像头视频

pangchaogit, 2019-02-22:


xianjimli, 2019-02-22:
应该可以。这个与具体硬件有关(我不太了解)，硬件一般都支持两层framebuffer，上面一层用作GUI，下面一层用作摄像头。把窗口使用透明背景色(对于565的格式，可能还需把某种颜色设置为透明色)，这样就可以两层进行混合了。

pangchaogit, 2019-02-25:
awtk下的现有控件支不支持在x86下模拟调用摄像头并显示视频，如果不支持，是否有一些底层接口去接收图片文件流并显示呢？

xianjimli, 2019-02-28:
我加了mutable_image控件，可以实现这个功能，你有空试试。

文档：https://github.com/zlgopen/awtk/blob/master/docs/manual/mutable_image_t.md

示例：https://github.com/zlgopen/awtk-c-demos/blob/master/demos/mutable_image.c

我研究一下ffmpeg，看能不能弄个更简单的控件。

erniuli, 2019-03-01:
你好，mutable_image_set_framebuffer能否提供一个简单的示例呢？

xianjimli, 2019-03-01:
没有硬件支持，不太好写例子。你把下面一层的framebffer相关参数设置进去即可。

pangchaogit, 2019-03-01:
你好。我参照v4l2+frambuffer  的方式获取到了摄像头下的视频，采用接口 mutable_image_set_framebuffer(image,width,high,format,(uint8_t*)fbp); fbp是frambuffer指针，format设置是  format=BITMAP_FMT_RGB565;程序运行后窗口没有图像显示，终端打印输出：mutable_image_prepare_image:31 mutable_image->prepare_image != NULL
mutable_image_on_paint_self:60 bitmap != NULL
theme_find_style:89 t != NULL && t->data != NULL
theme_find_style:89 t != NULL && t->data != NULL
这是我哪里没设置对么

xianjimli, 2019-03-01:
硬件不支持多层时，不要调用mutable_image_set_framebuffer。写个回调函数准备图片，调用mutable_image_set_prepare_image设置给mutable image即可。

pangchaogit, 2019-03-01:
现在改成在回调函数里面准备图片，设置s_bg_color，
static ret_t dummy_prepare_image(void* ctx, bitmap_t* image) {
    video_capture(src_image);
    //draw_framebuffer(src_image);
    rect_t r = rect_init(0, 0, image->w, image->h);
    image_fill(image, &r, s_bg_color);
    for(int i=0;i<IM_HEIGHT*IM_WIDTH;i++){
        s_bg_color.rgba.r = src_image[3*i+2];
        s_bg_color.rgba.g = src_image[3*i+1];
        s_bg_color.rgba.b = src_image[3*i];
    }

    return RET_OK;
}

mutable_image_set_prepare_image运行后屏幕显示花的图片是这样的，
[image]

xianjimli, 2019-03-01:
代码有问题。你需要把src_image里的数据，转换到image->data中去。不要着急，静下心去看看代码吧。

pangchaogit, 2019-03-04:
非常感谢，摄像头调用显示视频这功能可以显示。

xianjimli, 2019-03-04:
好的。

closed 2019-03-30 by xianjimli