zlgopen/awtk #90: 移植 awtk 遇到的一些问题

qgyhd1234, 2019-02-28:
我想把 awtk 移植到我的 stm32f429 开发板上，开发板运行了 rt-thread 系统。
我参考的是 [嵌入式平台移植注意事项](https://github.com/zlgopen/awtk/blob/master/docs/porting_common.md)
有些疑惑，想请官方人员解答下：
- 文档中要求加入工程的文件有 src/awtk.c，在最新版的源码中没找到
- SDL_COMPILE_TIME_ASSERT(enum, sizeof(SDL_DUMMY_ENUM) == sizeof(int)); 这句代码报错，不知道改怎么修改
- 编译时提示缺少 SDL_CreateTexture 等 SDL_ 开头的函数，分别是 lcd_sdl2c 和 main_loop_sdl_fb.c 两个文件调用的，这些函数是起到什么作用那？
- 我现在的 LCD 驱动实现是给上层 gui 一个 buffer，gui 只要把要显示的一帧画面画在 buff 里，然后调用 update 命令就好。这种方式是用对接到 AWTK 的哪些 API 上那？

xianjimli, 2019-03-01:
SDL是PC上用的。你参考429裸系统的移植吧：https://github.com/zlgopen/awtk-stm32f429igtx-raw

xianjimli, 2019-03-01:
我生成了一个STM32F429的文件列表，希望对你有帮助: https://github.com/zlgopen/awtk-stm32f429igtx-raw/blob/master/docs/files_list.md

qgyhd1234, 2019-03-01:
非常感谢，今晚回去试试

qgyhd1234, 2019-03-01:
@xianjimli 您好，参考 429 裸机工程后，我完成了我的移植工作，编译没有错误，但是在链接的时候提示我空间不够，我的 429 是 2M 的 FLASH，但是还是有很多的文件链接不进去，说明我加的功能太多了。我想问下有哪些功能是可以裁剪的吗？

我现在在 scons 脚本中添加了这些文件：
```
sources	= Glob('src/tkc/*.c') +\
  Glob('src/base/*.c') + \
  Glob('src/layouters/*.c') + \
  Glob('src/widgets/*.c') + \
  Glob('src/ui_loader/*.c') + \
  Glob('src/xml/*.c') + \
  Glob('src/svg/*.c') + \
  Glob('src/clip_board/*.c') + \
  Glob('src/font_loader/*.c') + \
  Glob('src/blend/*.c') + \
  Glob('src/ext_widgets/*.c') + \
  Glob('src/ext_widgets/*/*.c') + \
  Glob('src/image_loader/*.c') + \
  Glob('src/designer_support/*.c') + \
  Glob('src/widget_animators/*.c') + \
  Glob('src/platforms/pc/*.c') + \
  ['src/awtk_global.c'];

sources += ['src/main_loop/main_loop_simple.c'];
sources += ['src/window_animators/window_animator_fb.c'];
sources += ['src/input_methods/input_method_creator.c'] + Glob('src/lcd/lcd_mem_*.c') ;
sources += ['src/input_engines/input_engine_null.c']
sources +=['src/vgcanvas/vgcanvas_nanovg_soft.c'];
sources +=Glob('3rd/agge/agge/*.cpp');
sources +=Glob('3rd/nanovg/base/*.c');
sources +=Glob('3rd/nanovg/agge/*.cpp');
sources +=Glob('3rd/libunibreak/*.c');
```

xianjimli, 2019-03-02:
用的什么编译器？

qgyhd1234, 2019-03-02:
用的是 MDK，O2优化

qgyhd1234, 2019-03-02:
问题找到了，是 awtk_config.h里的配置问题，我用awtk里自带的例程编出来体积很大，用原子的那份就没问题，下进去屏幕点亮了，但是我的 LCD 驱动和 awtk 有点不太兼容，还得优化优化。

[image]

qgyhd1234, 2019-03-04:
@xianjimli 您好，之前移植完成后，开机会加载五个内容后进入桌面。昨天我重新生成了下工程，编译后卡在开机的 loading..... 不动了，这种情况该检查哪里

qgyhd1234, 2019-03-05:
找到问题了，我没开rtc，开启后就不会卡加载界面了，不知道为什么，第一次移植好后，没开rtc竟然也过了加载界面，就是比较卡。

qgyhd1234, 2019-03-05:
之前是开了个线程用来定时调用刷新命令，测试发现即使是 lcd_mem_bgr565_create_single_fb，lcd_device->swap 的回调函数也是起作用的，已经将刷新命令在 swap 函数里做调用

qgyhd1234, 2019-03-05:
@xianjimli 您好，我刚移植了下我显示屏的触摸驱动，现在测试发现，切换屏幕的时候的滑动效果比较差，像ppt一样，但是在调色盘，选择rgb色的时候拖动很流畅，这个是我的lcd驱动没有适配好的原因还是本身429就这么卡？

xianjimli, 2019-03-06:
429上跑800x480的屏不卡。看看这个优化文档：https://github.com/zlgopen/awtk/blob/master/docs/optimation.md

xianjimli, 2019-03-06:
最好使用双缓存。

closed 2019-05-14 by xianjimli