zlgopen/awtk #96: 如何构建支持xp的库

xinlengyuer, 2019-03-16:
如题，我才接触scons，不是很懂，我想编译一个支持xp的x86库，但是默认编译出来的是x64的不支持xp的库，我用的vs2017

xianjimli, 2019-03-18:
目前还不支持32位windows，我们想想办法，看能不能搞定。

soyzhc, 2019-03-19:
目前可以做到在win7 上编译awtk工程，编译的exe可以在winxp或32bit windows系统运行

要编译xp下运行的exe需要做以下修改

[awtk_support_winxp.diff.txt](https://github.com/zlgopen/awtk/files/2982595/awtk_support_winxp.diff.txt)
[awtk_tools_linkflags.diff.txt](https://github.com/zlgopen/awtk/files/2982597/awtk_tools_linkflags.diff.txt)

需要注意的是，修改后，exe会动态链接到crtdll（原版awtk是静态链接），因此需要在操作系统安装对应的vc运行时库，比如 MSVBCRT_AIO_2016.10.10_X86 X64.exe

> 使用crtdll的原因：经过测试发现，如果用vs2015编译静态链接crt的并兼容xp的exe，在xp系统执行stat等函数时会出错，导致awtk资源文件加载失败，可能是微软的bug，因为跟踪stat的实现发现其调用了xp不支持的API（GetFileInformationByHandleEx）

xianjimli, 2019-03-20:
已合并。谢谢。请参考：
https://github.com/zlgopen/awtk/blob/master/docs/how_to_build_windows_32bit_version.md

closed 2019-03-30 by xianjimli