zlgopen/awtk #126: 在aworks平台编译时，enums.c有些枚举值没有定义

soyzhc, 2019-05-27:
编译时出现错误
enums.c  line 211 ~ 331
比如
```
     {"PRINTSCREEN", 0, TK_KEY_PRINTSCREEN},
```
原因是keys.h这些枚举只在SDL2的情况下有定义，但aworks或其他裸机情况是没有SDL2的
```
#ifdef SDL2
TK_KEY_PRINTSCREEN = SDL_SCANCODE_TO_KEYCODE(SDL_SCANCODE_PRINTSCREEN),
...
```

还有一个问题是enums.c中keys_type_name_value部分按键重复定义了,比如
```
    {"PAGEUP", 0, TK_KEY_PAGEUP},
    {"PAGEDOWN", 0, TK_KEY_PAGEDOWN},
    {"F1", 0, TK_KEY_F1},
    {"F2", 0, TK_KEY_F2},
    {"F3", 0, TK_KEY_F3},
    {"F4", 0, TK_KEY_F4},
    {"F5", 0, TK_KEY_F5},
    {"F6", 0, TK_KEY_F6},
    {"F7", 0, TK_KEY_F7},
    {"F8", 0, TK_KEY_F8},
    {"F9", 0, TK_KEY_F9},
    {"F10", 0, TK_KEY_F10},
    {"F11", 0, TK_KEY_F11},
    {"F12", 0, TK_KEY_F12}
```

soyzhc, 2019-05-27:
修改后编译不过
src\base\enums.c(23): fatal error C1083: 无法打开包括文件: “base/utils.h”: No
such file or directory

xianjimli, 2019-05-28:
不好意思。麻烦再试试。

closed 2019-05-28 by soyzhc