zlgopen/awtk #129: VS2015编译awtk问题

ytchw101, 2019-05-27:
下载VS2015编译awtk 提示：  （按照文档依次下载安装python2.7 、scons 和vs2015）
  IndexError: list index out of range:
  File "C:\Users\Administrator\Desktop\awtk\SConstruct", line 11:
    OS_SUBSYSTEM_WINDOWS=awtk.OS_SUBSYSTEM_WINDOWS
  File "C:\Python27\scons\SCons\Defaults.py", line 88:
    _default_env = SCons.Environment.Environment(*args, **kw)
  File "C:\Python27\scons\SCons\Environment.py", line 982:
    apply_tools(self, tools, toolpath)
  File "C:\Python27\scons\SCons\Environment.py", line 107:
    env.Tool(tool)
  File "C:\Python27\scons\SCons\Environment.py", line 1789:
    tool(self)
  File "C:\Python27\scons\SCons\Tool\__init__.py", line 296:
    self.generate(env, *args, **kw)
  File "C:\Python27\scons\SCons\Tool\default.py", line 40:
    for t in SCons.Tool.tool_list(env['PLATFORM'], env):
  File "C:\Python27\scons\SCons\Tool\__init__.py", line 1266:

soyzhc, 2019-05-27:
是什么操作系统呢, Win7 x64 还是 Win10 x64

ytchw101, 2019-05-28:
win7 64

soyzhc, 2019-05-28:
安装的版本看看，我们这边没有碰到类似的问题
python 2.7.14
scons 3.0.1

ytchw101, 2019-05-28:
先说下我的步骤吧  看看哪里不对：
1.下载python-2.7.13.amd64.msi 安装后命令行输入python -V提示 Python 2.7.13   应该安装好了.
2.下载scons-3.0.4.zip解压缩后，CMD下进入解压后的scons-2.5.0文件夹然后执行命令 python setup.py install.
提示如下：
E:\软件安装包\scons-3.0.4>python setup.py install
running install
running build
running build_py
running build_scripts
running install_lib
running install_scripts
copying script/scons -> C:\Python27\Scripts\scons-3.0.4.py
copying script/scons -> C:\Python27\Scripts\scons.py
copying script/sconsign -> C:\Python27\Scripts\sconsign-3.0.4.py
copying script/sconsign -> C:\Python27\Scripts\sconsign.py
copying script/scons-time -> C:\Python27\Scripts\scons-time-3.0.4.py
copying script/scons-time -> C:\Python27\Scripts\scons-time.py
copying script/scons-configure-cache -> C:\Python27\Scripts\scons-configure-cach
e-3.0.4.py
copying script/scons-configure-cache -> C:\Python27\Scripts\scons-configure-cach
e.py
copying script/scons.bat -> C:\Python27
copying script/scons.bat -> C:\Python27\scons-3.0.4.bat
running install_data
running install_egg_info
Removing C:\Python27\scons\scons-3.0.4-py2.7.egg-info
Writing C:\Python27\scons\scons-3.0.4-py2.7.egg-info
Installed SCons library modules into C:\Python27\scons
Installed SCons scripts into C:\Python27\Scripts

3.下载VS2015安装, 选择自定义安装 勾选 Visual C++ 直到安装完成.
4.cmd下进入到下载的AWYK源码目录 即awtk-master文件夹下 输入命令 scons 提示如下

C:\Users\Administrator\Desktop\awtk>scons
scons: Reading SConscript files ...
TK_ROOT: C:\Users\Administrator\Desktop\awtk
IndexError: list index out of range:
  File "C:\Users\Administrator\Desktop\awtk\SConstruct", line 11
    OS_SUBSYSTEM_WINDOWS=awtk.OS_SUBSYSTEM_WINDOWS
  File "C:\Python27\scons\SCons\Defaults.py", line 88:
    _default_env = SCons.Environment.Environment(*args, **kw)
  File "C:\Python27\scons\SCons\Environment.py", line 982:
    apply_tools(self, tools, toolpath)
  File "C:\Python27\scons\SCons\Environment.py", line 107:
    env.Tool(tool)
  File "C:\Python27\scons\SCons\Environment.py", line 1789:
    tool(self)
  File "C:\Python27\scons\SCons\Tool\__init__.py", line 296:
    self.generate(env, *args, **kw)
  File "C:\Python27\scons\SCons\Tool\default.py", line 40:
    for t in SCons.Tool.tool_list(env['PLATFORM'], env):
  File "C:\Python27\scons\SCons\Tool\__init__.py", line 1266:
    c_compiler = FindTool(c_compilers, env) or c_compilers[0]
  File "C:\Python27\scons\SCons\Tool\__init__.py", line 1169:
    if t.exists(env):
  File "C:\Python27\scons\SCons\Tool\msvc.py", line 292:
    return msvc_exists(env)
  File "C:\Python27\scons\SCons\Tool\MSCommon\vc.py", line 759:
    vcs = cached_get_installed_vcs(env)
  File "C:\Python27\scons\SCons\Tool\MSCommon\vc.py", line 524:
    ret = get_installed_vcs(env)
  File "C:\Python27\scons\SCons\Tool\MSCommon\vc.py", line 535:
    VC_DIR = find_vc_pdir(ver)
  File "C:\Python27\scons\SCons\Tool\MSCommon\vc.py", line 344:
    comps = find_vc_pdir_vswhere(msvc_version)
  File "C:\Python27\scons\SCons\Tool\MSCommon\vc.py", line 306:
    vc_pdir = os.path.join(vsdir[0], 'VC')

注：系统是win7 64位   软件都是装在C盘  ，源码放在桌面文件夹名字我改成 awtk

xianjimli, 2019-05-28:
scons-3.0.4 可能需要python3

ytchw101, 2019-05-28:
重新下载安装python3.7.3 后  运行 还是这个提示

C:\Users\Administrator\Desktop\awtk-master>scons
scons: Reading SConscript files ...
TK_ROOT: C:\Users\Administrator\Desktop\awtk-master
IndexError: list index out of range:
  File "C:\Users\Administrator\Desktop\awtk-master\SConstruct", line 11:
    OS_SUBSYSTEM_WINDOWS=awtk.OS_SUBSYSTEM_WINDOWS
  File "C:\Python37\scons\SCons\Defaults.py", line 88:
    _default_env = SCons.Environment.Environment(*args, **kw)
  File "C:\Python37\scons\SCons\Environment.py", line 982:
    apply_tools(self, tools, toolpath)
  File "C:\Python37\scons\SCons\Environment.py", line 107:
    env.Tool(tool)
  File "C:\Python37\scons\SCons\Environment.py", line 1789:
    tool(self)
  File "C:\Python37\scons\SCons\Tool\__init__.py", line 296:
    self.generate(env, *args, **kw)
  File "C:\Python37\scons\SCons\Tool\default.py", line 40:
    for t in SCons.Tool.tool_list(env['PLATFORM'], env):
  File "C:\Python37\scons\SCons\Tool\__init__.py", line 1266:
    c_compiler = FindTool(c_compilers, env) or c_compilers[0]
  File "C:\Python37\scons\SCons\Tool\__init__.py", line 1169:
    if t.exists(env):
  File "C:\Python37\scons\SCons\Tool\msvc.py", line 292:
    return msvc_exists(env)
  File "C:\Python37\scons\SCons\Tool\MSCommon\vc.py", line 759:
    vcs = cached_get_installed_vcs(env)
  File "C:\Python37\scons\SCons\Tool\MSCommon\vc.py", line 524:
    ret = get_installed_vcs(env)
  File "C:\Python37\scons\SCons\Tool\MSCommon\vc.py", line 535:
    VC_DIR = find_vc_pdir(ver)
  File "C:\Python37\scons\SCons\Tool\MSCommon\vc.py", line 344:
    comps = find_vc_pdir_vswhere(msvc_version)
  File "C:\Python37\scons\SCons\Tool\MSCommon\vc.py", line 306:
    vc_pdir = os.path.join(vsdir[0], 'VC')

请问下：
1.第一句指向File "C:\Users\Administrator\Desktop\awtk\SConstruct", line 11 ，  SConstruct文件有需要修改什么吗？（源码awtk-master是我前段时间下载的没动过）
2.现在电脑安装的vs2015 之前好像在哪里看到过要装vs2017,和这个没关系吧？ 
 另外 运行scons和我装vs是有什么关系？

ytchw101, 2019-05-28:
问题找到了 又重新下载最新版本scons-3.0.5 安装后  再cmd 编译awtk-master源码 可以了 最终提示scons: done building targets.
但按照文档生成.data又有个问题

1.下载awtk-examples-master.zip文件在桌面解压,同时源码文件awtk-master(之前scons编译好的)也是放在桌面.
2.修改awtk-examples-master/Meter-Demo目录下assets_gen.bat文件中@set AWTK_DIR=../../awtk-master 
即如下：
@rem 这里的AWTK_DIR的值在SConstruct脚本的getAwtkRoot()函数会用到，修改时请注意

@set AWTK_DIR=../../awtk-master

cd res
call assets_gen.bat ../%AWTK_DIR%/bin
cd ..

3.其他文件都没动，然后cmd下运行assets_gen.bat文件 提示：
C:\Users\Administrator\Desktop\awtk-examples-master\Meter-Demo>assets_gen.bat
C:\Users\Administrator\Desktop\awtk-examples-master\Meter-Demo>cd res
C:\Users\Administrator\Desktop\awtk-examples-master\Meter-Demo\res>call assets_g
en.bat ../../../awtk-master/bin
C:\Users\Administrator\Desktop\awtk-examples-master\Meter-Demo\res>py -2 ../asse
ts_gen.py ../../../awtk-master/bin ./assets/raw ./assets/inc
'py' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
C:\Users\Administrator\Desktop\awtk-examples-master\Meter-Demo\res>py -2 ../asse
ts_c_gen.py ./assets
'py' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
C:\Users\Administrator\Desktop\awtk-examples-master\Meter-Demo\res>cd ..
C:\Users\Administrator\Desktop\awtk-examples-master\Meter-Demo>

这是啥意思啊？

ytchw101, 2019-05-28:
是不是\Meter-Demo\res目录下的assets_gen.bat里面也要修改？
下面的我修改了AWTK_BIN_DIR目录，关于python的怎么改？ python现在我装的是python3.7.3 

@set AWTK_BIN_DIR=C:\Users\Administrator\Desktop\awtk-master\bin
@set PY_VERSION=python
@if exist %SystemRoot%\py.exe (
	@set PY_VERSION=py -2
)

@rem 为了适配awtk版本(sha=4b7f6c59449fab21dd7859e5ace6422533dddb82)
@if exist %cd%\assets\raw\fonts\default.mini.ttf (
    @rename %cd%\assets\raw\fonts\default.mini.ttf default_mini.ttf
)

%PY_VERSION% ../assets_gen.py %AWTK_BIN_DIR% ./assets/raw ./assets/inc
%PY_VERSION% ../assets_c_gen.py ./assets

closed 2019-05-29 by ytchw101