zlgopen/awtk #131: button、label、dialog等可以做半透明吗？怎么实现？dialog能嵌套在window中吗？

YinWenshuo, 2019-05-29:
两个问题，
1、button、label、dialog等可以做半透明吗？如果可以做的话，在c代码里面或者xml里面怎么实现？最好能在xml里面实现，更加直观。
2、在xml中，dialog能嵌套在window中吗，像button、label一样。如下：
<window theme="main">
    <view x="400" y="0" w="770" h="11%">
	<dialog anim_hint="slide_down"  x="400" y="0" w="770" h="11%" text="abc!" style="s_abc">
    </view>

    <view  x="1170" y="top" w="110" h="100%">
        <view  x="0" y="top" w="100%" h="20%" layout="r:1 c:1">     
            <label name="swc_progress" style="s_swc_icon"/>
	</view>
	<view  x="0" y="20%" w="100%" h="80%" layout="r:4 c:1">    
            <button name="btn_a" style="s_a"/>
            <button name="btn_b" style="s_b"/>
	    <button name="btn_c" style="s_c"/>
            <button name="btn_d" style="s_d"/>
        </view>    
    </view>
</window>

YinWenshuo, 2019-05-29:
第2个问题，刚才试了下，xml中，window里面不能嵌dialog，不知道是不是我上面的写法不对，还是说不支持。

xianjimli, 2019-05-29:
1.dialog不能嵌入到window中。
2.半透效果在style里把背景颜色指定为半透即可。如：
```
<normal     bg_color="rgba(200,200,200,0.1)" />
```
3.半透的dialog不能支持窗口动画。

YinWenshuo, 2019-05-29:
好的，多谢回复。例子和文档里面好像都没有这样写， bg_color="rgba(200,200,200,0.1)"，要不是您指导下，我要摸索好久，哈哈，谢谢。

YinWenshuo, 2019-05-29:
试了一下，这样是可以半透明的，但是只对用颜色画的label或者button起作用，对图片好像不起作用？如下：
<style name="s_a" icon="a_btn_n" >
        <normal bg_color="rgba(200,200,200,0.6)" />
        <pressed icon="a_btn_p" />
        <over       fg_color="#e0e0e0" />
        <disable    fg_color="gray" text_color="#d0d0d0" />
    </style>

xianjimli, 2019-05-30:
增加了一些文档，你有空看看：

* 增加了颜色格式的文档：
https://github.com/zlgopen/awtk/blob/master/docs/color_format.md

* 增加了关于半透明的问答：
https://github.com/zlgopen/awtk/blob/master/docs/faq.md

xianjimli, 2019-05-30:
在嵌入式平台，半透明效果会使性能大幅下降，请谨慎使用。

YinWenshuo, 2019-05-30:
好的，多谢。

closed 2019-06-04 by YinWenshuo