zlgopen/awtk #132: 嵌入式添加中文输入法

ytchw101, 2019-05-29:
嵌入式平台(ARM9),按照文档7.4节添加 加入中文输入法所需的文件到工程，最后一步是“修改 awtk\SConstruct 文件” 。 我打开我这个SConstruct 文件和说明里格式不太一样 ，然后我把我这个SConstruct文件里没有的（如下）添加到里面。

#INPUT_ENGINE='null'
INPUT_ENGINE='pinyin'

CPPPATH=[TK_ROOT,
TK_SRC,
TK_3RD_ROOT,
joinPath(TK_SRC, 'ext_widgets'),
joinPath(TK_3RD_ROOT, 'bgfx/bgfx/include'),
joinPath(TK_3RD_ROOT, 'bgfx/bx/include'),
joinPath(TK_3RD_ROOT, 'bgfx/bimg/include'),
joinPath(TK_3RD_ROOT, 'agge'),
joinPath(TK_3RD_ROOT, 'agg/include'),
joinPath(TK_3RD_ROOT, 'nanovg'),
joinPath(TK_3RD_ROOT, 'nanovg/gl'),
joinPath(TK_3RD_ROOT, 'nanovg/base'),
joinPath(TK_3RD_ROOT, 'nanovg/agge'),
joinPath(TK_3RD_ROOT, 'nanovg/bgfx'),
joinPath(TK_3RD_ROOT, 'SDL/src'),
joinPath(TK_3RD_ROOT, 'SDL/include'),
joinPath(TK_3RD_ROOT, 'agge/src'),
joinPath(TK_3RD_ROOT, 'agge/include'),
joinPath(TK_3RD_ROOT, 'gpinyin/include'),
joinPath(TK_3RD_ROOT, 'libunibreak'),
TK_TOOLS_ROOT] + OS_CPPPATH

操作做完，（工程里添加的是stm32429igtx的例子），运行程序后 打开里面的KeyBoard  切换到中文输入 输入还是显示英文字符  没有显示中文？  是我哪地方弄不对 还是哪里还需要做什么操作？？？

另工程配置中没有定义 不使用输入法 WITH_NULL_IM

xianjimli, 2019-05-29:
嵌入式linux版本吗？

ytchw101, 2019-05-29:
裸跑的     问下那个SConstruct文件 按照我上面那个改  对不对？？？

xianjimli, 2019-05-29:
我不知道怎么用scons编译裸跑的版本。搞定之后，帮忙写篇文档:)

ytchw101, 2019-05-30:
现在情况是打开KeyBoard 控件切换到中文  即“中” 不管按什么字符都是英文字母  ，切换到英文即 “En” ，有几个字母按下会出汉字  比如：按s 出来 “”试“ 按 d 出来 ”定“ 按 ”w“出来 ”文“  其他很多字母怎么按都不出汉字的 ？ 代码运行的是demo里的 application_init();      这种情况不知道是哪里没配置好 还是啥问题？

xianjimli, 2019-05-30:
应该字体的问题，换一个完整的字体试试。

ytchw101, 2019-05-30:
输入法里用的字体数据是gpinyin.inc  里的 data_gpinyin 数组（1M左右）数据吧？   意思是把这个换了吗？

xianjimli, 2019-05-30:
fonts目录下的tff文件才是字体。

ytchw101, 2019-05-30:
如果我换个.ttf矢量字体    比如MYFONT.ttf    那么是先用assets_gen.bat生成.res资源 然后程序中 在assets_init 里再加载吧？        输入法中用到的字库数据应该是 data_gpinyin 数组数据吧？？？？

xianjimli, 2019-05-30:
把default_full.ttf拷贝到default.ttf即可，重新生成资源并编译即可。

ytchw101, 2019-05-30:
那gpinyin.inc文件 不是给输入法用的吗？ 这个文件怎么更改  ...

xianjimli, 2019-05-30:
不用改啊。

ytchw101, 2019-05-31:
这个 gpinyin.inc 是做什么用的？

xianjimli, 2019-05-31:
拼音数据。

ytchw101, 2019-05-31:
现在的方式是：1.使用完整字体(也就是default.ttf或其他完整字体的.ttf文件都可以)  2.默认使用stb 加载矢量字体  （也就是宏#define WITH_STB_FONT  1  打开） 现在输入中文可以了  偶尔输入中文不出来 还没找到原因！！          
现在又遇到个问题是 我改成用freetype加载矢量字体 (也就是打开  #define WITH_FT_FONT   1 宏， 关闭 WITH_STB_FONT)  出现有时还没进到界面直接程序卡死了  有时进去了 里面显示的汉字乱码了一团一团的  不知道  freetype方式加载 有没其他什么要改或者要配置的 要注意地方？  或者是我程序原因， 程序上我也在找原因

closed 2019-08-13 by ytchw101

pqlcyl, 2020-04-23:
你好！请问中文输入法候选字中文无法显示原因是什么？我现在也出现这个问题，望指导，谢谢！

pqlcyl, 2020-04-23:
解决了，和字库有关系

ytchw101, 2020-04-23:
需要加载个字库资源