zlgopen/awtk #142: freetype支持truetype字体显示问题

ytchw101, 2019-06-03:
使用freetype方式（即配置宏定义#define WITH_FT_FONT   1） 加载字体 为什么显示的字都重叠在一起     配置stb方式（即 #define WITH_STB_FONT 1）显示正常。   

freetype方式  还有其他什么要作更改吗？

资源加载函数有个这句话 ：
#if defined(WITH_MINI_FONT) && (defined(WITH_STB_FONT) || defined(WITH_FT_FONT))
assets_manager_add(rm, font_default_mini);  

那么WITH_FT_FONT方式下不能加载default.ttf等完整字体吗？ 我现在默认上面这句话注释掉了  我加载的是default.ttf 完整字体 这样 输入法 汉字才能都显示出来

xianjimli, 2019-06-03:
在MacOs下测试没问题啊，修改一下awtk_config.py重新编译就行了：

```
 65 #COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_STB_FONT '
 67 COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_FT_FONT '
```

ytchw101, 2019-06-04:
下载最新源码 awtk-master 按照上面修改awtk_config.py
#COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_STB_FONT '
COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_FT_FONT '

后编译源码（windows7 64位系统）提示错误：
font_loader_ft.c
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25356): error C2236: 意外的标记“struct”。是否忘记了“;”?
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25356): error C2332: “struct”: 缺少标记名
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25383): error C2332: “struct”: 缺少标记名
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25383): error C2513: “”: 在“=”前没有声明变量
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25385): error C2332: “struct”: 缺少标记名
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25385): error C2059: 语法错误:“类型”
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25391): error C2332: “struct”: 缺少标记名
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25391): error C2059: 语法错误:“类型”
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25392): error C2059: 语法错误:“if”
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25394): error C2059: 语法错误:“else”
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25396): error C2059: 语法错误:“if”
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25402): error C2059: 语法错误:“return”
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(25404): error C2059: 语法错误:“}”
C:\Users\Administrator\Desktop\awtk-master\3rd\freetype_single_header/freetype.h
(116108): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件
保存为 Unicode 格式以防止数据丢失
scons: *** [src\font_loader\font_loader_ft.obj] Error 2
scons: building terminated because of errors.

注意：如果改回去即改成
COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_STB_FONT '
#COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_FT_FONT '

编译是可以成功的。

xianjimli, 2019-06-04:
看了一下，VC不支持单文件的freetype.h(前面的代码使用了后面的结构定义)。如果在windows下使用freetype，需要自己编译freetype库。

closed 2019-07-05 by ytchw101

poweel, 2019-08-02:
去掉下面几个函数（25347行-25432行），就可以在windows上编译通过了。
ft_property_do
FT_Property_Get
FT_Property_Set