zlgopen/awtk #143: 请教freetype解析文字

ytchw101, 2019-06-03:
我现在包含freetype.h头文件用freetype解析文字能得到文字 子模数据和信息  但是改变字体大小 他使用的好像还是style里设置的字体大小， 我该怎么做。也就是下面这句话怎样才有效
  int font_size = 64;
//			FT_Set_Pixel_Sizes(m_face, font_size, 0);
	FT_Error res = FT_Set_Char_Size(m_face, 0, font_size * 64, 0, 50);
注意：
函数代码如下：
void FreeType_LoadChar(const wchar_t str)
{
	#if 1
	int bErr = FT_Init_FreeType(&m_library);
	if(bErr)
	{
		 rt_kprintf("FT_Init_FreeType error!\r\n");
	}
//  bErr = FT_New_Face(m_library,"C:\\WINDOWS\\Fonts\\simkai.ttf",0,&m_face);//simhei
	 uint32_t i = 0;
	 uint32_t nr = assets_manager()->assets.size;
	 const asset_info_t** all = (const asset_info_t**)(assets_manager()->assets.elms);
   const asset_info_t* iter = NULL;
	
	 for (i = 0; i < nr; i++) 
	 {
			if(all[i]->type == ASSET_TYPE_FONT)
			{
				 iter = all[i];
				 break;
			}
	 }
	
	if (FT_New_Memory_Face(m_library, iter->data, iter->size, 0, &m_face)) 
	{
     FT_Done_FreeType(m_library);
		 rt_kprintf("FT_New_Memory_Face error!\r\n");
  }
	else
	{

	}
		
//	FT_Select_Charmap(m_face,FT_ENCODING_UNICODE);
	
	if (FT_Select_Charmap(m_face, FT_ENCODING_UNICODE)) {
//    FT_Select_Charmap(m_face, FT_ENCODING_ADOBE_LATIN_1);
  }
	
  int font_size = 64;
//			FT_Set_Pixel_Sizes(m_face, font_size, 0);
	FT_Error res = FT_Set_Char_Size(m_face, 0, font_size * 64, 0, 50);
	rt_kprintf("FT_Set_Char_Size  res=%d!\r\n",res);
	int ret = FT_Load_Char(m_face, str, FT_LOAD_DEFAULT | FT_LOAD_RENDER); 
	
	if(ret != 0)
	{
		 rt_kprintf("FT_Load_Char error!\r\n");
	}
	
	 rt_kprintf("w:%d,h:%d\r\n\r\n",m_face->glyph->bitmap.width,m_face->glyph->bitmap.rows);
	
	 int k=0, j=0;
   int height = m_face->glyph->bitmap.rows;
   int wid = m_face->glyph->bitmap.width;
   unsigned char*  buffer;
   buffer = m_face->glyph->bitmap.buffer;

   for(j = 0; j < height; j++)
   {
			for(k = 0; k < wid ; k++)
			{
					if(buffer[k + wid*j] != 0x00)
					{
							rt_kprintf("@");
					}
					else
					{
							rt_kprintf("-");
					}
			}
			rt_kprintf("\r\n");
   }
	 
	 #endif
}

xianjimli, 2019-06-04:
新加了inline style，你可以试试：https://github.com/zlgopen/awtk/blob/master/docs/theme.md

```
widget_set_style_int(b, "font_size", 24);
```

ytchw101, 2019-06-04:
哦   这个是动态改某个widget 字体大小 ！！ 我现在按照上面的步骤解析某个汉字字模  动态改字体大小 再解析但 发现使用 
       FT_Set_Pixel_Sizes(m_face, font_size, 0);或者
	FT_Error res = FT_Set_Char_Size(m_face, 0, font_size * 64, 0, 50);

改都没反应   解析出来的字模数据默认大小还是32点的字体数据 (不知道他调用哪里的font_size)   请问  这我该怎么去做。

xianjimli, 2019-06-04:
看下freetype的文档

closed 2019-07-17 by ytchw101