zlgopen/awtk #144: 关于文本显示的问题请教（label和rich_text）

YinWenshuo, 2019-06-04:
我的需求是：在一个指定大小的区域内显示一些提示信息，提示信息比较长，而且要求显示信息要做多国语言显示。
我的问题是：使用label的话，因为做多国语言，翻译里面插入换行符不起作用，如下：
<language name="en_US">ABCDSFJJFFJFJFJFFKDSFFLSGGGGKDKGKGK  \n DJJDJDJHHFHDFHDFHDFDFJDFDFJDJFDJFDFKJ</language>
使用rich_text的话，好像没有看到怎么使用多语言（tr_text）,而且，rich_text的用法感觉看起来有点头大，比如例子里面：&quota是什么意思？
  <rich_text line_gap="5" x="0" y="210" w="100%" h="200" text="<image name=&quota;bricks&quota;/><font color=&quota;gold&quota; align_v=&quota;bottom&quota; size=&quota;24&quota;>hello awtk!</font><font color=&quota;green&quota; size=&quota;20&quota;>ProTip! The feed shows you events from people you follow and repositories you watch. 
hello world. </font><font color=&quota;red&quota; size=&quota;20&quota;>确定取消中文字符测试。确定。取消。中文字符测试。</font>" />

xianjimli, 2019-06-04:
&quota;表示双引号，这是XML的规范。这样确实很麻烦，我刚支持了把属性放到独立property标签中，把需要转义的字符串放到CDATA中即可。如：

```xml
  <rich_text x="0" y="0" w="100%" h="60">
    <property name="text">
      <![CDATA[<image name="bricks"/><font align_v="middle">hello awtk!</font><font color="red" size="32">ZLG</font>]]>
    </property>
  </rich_text>
```

详情请参考：https://github.com/zlgopen/awtk/blob/master/docs/ui_desc.md

xianjimli, 2019-06-04:
翻译的多行文本要这样写：
```xml
<string name="multi line text">
  <language name="en_US">
it
is
a
multi-line
Translate text
(center)
  </language>
</string>
```

YinWenshuo, 2019-06-05:
多谢回复，原来直接在翻译里面换行就可以，完美的解决了我的问题。谢谢。

closed 2019-06-06 by YinWenshuo