zlgopen/awtk #145: 显示内存位图

ytchw101, 2019-06-05:
不使用系统加载的图片资源显示图片。而是根据自己动态生成位图数据 显示出图片。

现在是创建一个画布控件 然后注册on_paint_event事件中绘制位图  这样可行吗？ 代码如下：

static ret_t on_paint_event(void* ctx, event_t* evt) 
{
   widget_t* canvas_widget = WIDGET(ctx);
   canvas_t* c = paint_event_cast(evt)->c;
   vgcanvas_t* vg = canvas_get_vgcanvas(c);
   color_t bg = color_init(0xe0, 0xe0, 0xe0, 0xff);
   color_t tc = color_init(0, 0, 0, 0xff);
   rect_t r = rect_init(canvas_widget->x, canvas_widget->y, canvas_widget->w, canvas_widget->h);
 
   vgcanvas_save(vg);
   vgcanvas_clip_rect(vg, r.x, r.y, r.w, r.h);
   vgcanvas_translate(vg, r.x, r.y);
   ret_t res = RET_FAIL;
		
    bitmap_t *img = bitmap_create_ex(11,22,22,BITMAP_FMT_RGB565);
		
    img->should_free_data = TRUE;
    img->should_free_handle = TRUE;

		if(img)
		{
			res = bitmap_init(img,11,22,BITMAP_FMT_RGB565, (uint8_t*)data_wen);//data_wen
                                                                                                                                        为位图数据
			rt_kprintf("bitmap_init res=%d\r\n", res);
			res = canvas_draw_icon(c, img, 0, 0);
			rt_kprintf("canvas_draw_icon res=%d\r\n", res);
		}
		else
		{
			 rt_kprintf("bitmap_create null!\r\n");
		}
		
   vgcanvas_restore(vg);
	
   return RET_OK;
}

注:每次 canvas_draw_icon  返回RET_NOT_IMPL  （没实现/不支持）   ，换成canvas_draw_image_at也是一样返回值。

请问根据位图数据动态显示位图   可不可以这样或者还有其他什么方法！？

xianjimli, 2019-06-05:
OpenGL不支持565格式。

请参考：src/ext_widgets/color_picker/color_component.c

ytchw101, 2019-06-05:
试了下改成   BITMAP_FMT_BGR565  可以显示了

closed 2019-06-06 by xianjimli