zlgopen/awtk #146: 发现一个bug（使用多语言时icon无法显示，文本无法按照代码意愿排列）

YinWenshuo, 2019-06-05:
问题详述：
我的xml代码如下
ui xml：
<window>
    <view x="400" y="0" w="770" h="70">
        <label name="tips1" x="center" y="m" w="100%" h="100%" tr_text="main_tips_safe" visible="FALSE" />
    </view>
</window>

style xml：
<label>
    <style name="default" icon="info" icon_at="left" text_color="black" text_align_h="center" text_align_v="middle" font_size="28">
        <normal bg_color="rgba(40,47,117,0.6)" />
    </style>
</label>

按照上面的写法，问题有两个，
1、我预期的info图标没有显示出来，
2、我的文本（即main_tips_safe在多语言翻译strings.xml中对应的字符串）在y方向上没有居中显示，而是靠上显示。

如果：
1、我把ui文件中的tr_text改成text，那么，我预期的info图标显示出来了，文本“main_tips_safe”在y方向上居中显示了，只是文本不是我要的文本。
2、或者我把ui文件中的 tr_text="main_tips_safe" 改成 tr_text="dddfdfdfdfdf" （dddfdfdfdfdf在翻译文件中不存在），那么，也可以显示info图标，且文本“dddfdfdfdfdf”在y方向上居中显示了

所以，这应该是个bug吧，我代码应该没问题。请帮忙debug。谢谢！

YinWenshuo, 2019-06-05:
对了，我在dialog_title中类似的用法没有这问题，可能应该是label的一个bug

xianjimli, 2019-06-05:
好的，谢谢。今天太忙，明天我看看。

YinWenshuo, 2019-06-05:
另外，再请教个问题，icon只能用icon_at指定left/right/top/bottom吗，能不能设定icon跟text之间的间距？

xianjimli, 2019-06-05:
> 另外，再请教个问题，icon只能用icon_at指定left/right/top/bottom吗，能不能设定icon跟text之间的间距？

用spacer试试。

YinWenshuo, 2019-06-06:
> > 另外，再请教个问题，icon只能用icon_at指定left/right/top/bottom吗，能不能设定icon跟text之间的间距？
> 
> 用spacer试试。

ok，谢谢。另外，再请教个问题，border可以设置border_color,能不能设定border的粗细啊？怎么设置？有时候border看起来太细了，不明显，想把border加粗。

xianjimli, 2019-06-06:
我这里没问题。style之间一定要写状态，否则无效。

``` xml
<style name="default" icon="info" icon_at="left" text_color="black" text_align_h="center" text_align_v="middle" font_size="28">
 <normal  />
</style>
```

xianjimli, 2019-06-06:
我加了border_width参数，你试试。

YinWenshuo, 2019-06-12:
> 我这里没问题。style之间一定要写状态，否则无效。
> 
> ```
> <style name="default" icon="info" icon_at="left" text_color="black" text_align_h="center" text_align_v="middle" font_size="28">
>  <normal  />
> </style>
> ```

恩，我的style里面有状态。
我知道我的问题在哪儿了，是 tr_text="main_tips_safe"在翻译strings.xml文件里面对应的翻译文本太长，我换行了，如果不换行就没有问题了。那么，可以不可以解决呢？

closed 2020-06-01 by YinWenshuo