zlgopen/awtk #154: 有没有截屏功能，截取指定范围的屏幕图像（x,y,w,h）

YinWenshuo, 2019-06-17:
如题
有没有截屏功能，截取指定范围的屏幕图像（x,y,w,h）

xianjimli, 2019-06-17:
好的，近期加上。

closed 2020-06-01 by YinWenshuo

suve7, 2022-02-14:
大佬这个功能添加了吗?

suve7, 2022-02-14:
这个功能添加了吗？怎么使用它？

xianjimli, 2022-02-14:
```c
/**
 * @method widget_take_snapshot_rect
 * 创建一个bitmap对象，将控件绘制到bitmap上并且可以设置该控件的截屏区域，返回bitmap对象。
 *
 * 调用者需要调用bitmap_destroy销毁返回的bitmap对象。
 *
 *```c
 * bitmap_t* bitmap = widget_take_snapshot_rect(window_manager(), NULL);
 * bitmap_save_png(bitmap, "test.png");
 * bitmap_destroy(bitmap);
 *``` 
 *
 * @param {widget_t*} widget 控件对象。
 * @param {const rect_t*} r 截屏区域（输入NULL，则为控件全区域截屏）。
 *
 * @return {bitmap_t*} 返回位图对象。
 */
bitmap_t* widget_take_snapshot_rect(widget_t* widget, const rect_t* r);
```

suve7, 2022-02-14:
> ```c
> /**
>  * @method widget_take_snapshot_rect
>  * 创建一个bitmap对象，将控件绘制到bitmap上并且可以设置该控件的截屏区域，返回bitmap对象。
>  *
>  * 调用者需要调用bitmap_destroy销毁返回的bitmap对象。
>  *
>  *```c
>  * bitmap_t* bitmap = [...]

好的，非常感谢！！