zlgopen/awtk #157: stb方式解析ttf字符

ytchw101, 2019-06-19:
就是想用stb方式解析字符得到字模数据？现在我用下面的代码  之前是能解析出来 现在不知知道什么原因 总是解析的不对  打印得到  ret=0,x:0,y:-128,w:3,h:192  advance=-1073512448

代码如下： 不知道能不能这样写  有什么需要注意的嘛？  之前代码 和现在感觉也就多加了点界面显示代码..      LoadChar没动过

extern font_t* m_ft; 
//m_ft为tk_add_font函数中font_truetype_create返回的font_t*  

ret_t LoadChar(const wchar_t str)
{
	 ret_t ret = RET_FAIL;
	 glyph_t* gy;
	 rt_kprintf("Loadchar ----------- name:%s\r\n", m_ft->name);
	 ret = font_my_get_glyph(m_ft, str,64,gy);  //
	 rt_kprintf("ret=%d,x:%d,y:%d,w:%d,h:%d  advance=%d\r\n", ret,gy->x,gy->y,gy->w,gy-
                             >h,gy->advance);
   int k=0, j=0;
   int height =gy->h;
   int wid = gy->w;
   const uint8_t*  buffer;
   buffer = gy->data;

   for(j = 0; j < height; j++)
   {
			for(k = 0; k < wid ; k++)
			{
					if(buffer[k + wid*j] != 0x00)
					{
							rt_kprintf("@");
					}
					else
					{
							rt_kprintf("-");
					}
			}
			rt_kprintf("\r\n");
   }
	 
//	 font_my_stb_destroy(m_ft);
	 
	 if(gy->data)
	 {
	    TKMEM_FREE(gy->data);
            gy->data = NULL;
           rt_kprintf("Free gy->data...\r\n");
	 }
   return RET_OK;
}

ytchw101, 2019-06-20:
现在换了个简单方式： （其中g_canvas为canvas_init()返回的指针）
   glyph_t gy;	
   font_size_t font_size = 64;
   return_value_if_fail(font_get_glyph(g_canvas->font, str, font_size, &gy) == RET_OK, RET_BAD_PARAMS);

这样也能解析出字符数据gy.data,  请问可以这样用吗？   其他的我实在是没看到什么方法

另外这个gy.data 不能释放吗 比如我使用下面：
	 if(gy.data)
	 {
	    TKMEM_FREE(gy.data);
        gy.data = NULL;
		  rt_kprintf("Free gy.data...\r\n");
	 }

释放了gy.data后 我界面本来有的和这个解析的字符一样的文字 再刷新就乱码了  这是什么原理？？？

xianjimli, 2019-06-20:
可以。不能释放，由缓存管理，自动释放。

xianjimli, 2019-06-20:
你要实现实现什么效果？

ytchw101, 2019-06-20:
哦 好的   就是获得任意大小字体字模数据

ytchw101, 2019-06-21:
另外请问一下 这个gy.data  你上面说是由缓存管理 自动释放 ，那么是过一定时间自己释放还是等到程序结束时才会释放 ？

xianjimli, 2019-06-24:
cache自动淘汰不用的字符。

ytchw101, 2019-06-25:
哦 谢谢

closed 2019-07-05 by ytchw101