zlgopen/awtk #161: widget_set_text请教

ytchw101, 2019-06-29:
请问这个函数对text长度有要求嘛？    我例子是创建一个label控件 然后widget_set_text  如果text内容很长我的界面就直接卡死  不知道怎么回事？

主要代码如下：

	widget_use_style(label, "label_printobj"); //自己定义的风格改变背景和字体颜色
	widget_set_style_int(label, "font_size", fontsize);	 //字体大小是50
	widget_set_text(label, text); //
	label_resize_to_content(label,0,0xffff,0,0xffff); //根据内容设置label大小

ytchw101, 2019-06-29:
多次测试：每次输入10个字不超过屏可显示大小， 显示上去 ，输入超过20次左右出现一次卡死 。  
现在我程序是一个界面建立了个mledit控件输入文字，输入好 点击按钮 响应on_btn_textedit_ok，切换到另一个界面并显示编辑的文字到label上。

static ret_t on_btn_textedit_ok(void* ctx, event_t* evt)
{
	widget_t* win = WIDGET(ctx); //编辑界面edit指针
	(void)evt;
	const wchar_t* text = widget_get_text(win); //获得输入的内容
	CreateLabel(text, 50, OBJ_CREATE_TEXT, win->isModify); //
	widget_set_prop_bool(page.par_main, WIDGET_PROP_VISIBLE, TRUE); //主界面可显示
	
       return window_close(win->parent); //关闭文字编辑界面
}

也不知道是不是我界面切换会导致卡死， 输入10个汉字可能需要输入创建很多个label才会出现卡死，但是输入字符较长的话 比如20  30 个汉字, 再去显示可能1 、 2次就会卡死。    

找了半天原因也没找到,  所以想请教下....

ytchw101, 2019-06-29:
经多次测试可能是 界面切换导致崩溃。   我在主界面搞个按钮按下发送30多个文字基本每次不会卡死，而在编辑界面编辑几十个汉字 然后 widget_set_text 再关闭编辑界面->再显示主界面  这样和容易卡死  而且字数越多或者界面有文字就很容易卡死       ！！

ytchw101, 2019-06-29:
有没有一个界面关闭前 往另外一个界面发送消息或通知的  这样函数  什么的？   显示界面如何在编辑界面编辑完后关闭前获得编辑的内容。

xianjimli, 2019-06-29:
> 有没有一个界面关闭前 往另外一个界面发送消息或通知的 这样函数 什么的？ 显示界面如何在编辑界面编辑完后关闭前获得编辑的内容。

注册窗口的EVT\_WINDOW\_CLOSE事件。

xianjimli, 2019-06-29:
> widget\_set\_text

是多行编辑器吗？是编辑过程中卡死，还是widget\_set\_text卡死

xianjimli, 2019-06-29:
不要着急，逐步简化，一步一步的定位问题。

xianjimli, 2019-07-01:
> win->isModify

isModify貌似不是widget的成员。

ytchw101, 2019-07-01:
> isModify貌似不是widget的成员。

 isModify是我自己添加的属性

ytchw101, 2019-07-01:
> > widget_set_text
> 
> 是多行编辑器吗？是编辑过程中卡死，还是widget_set_text卡死

都不是，是编辑界面编辑完后，点击确定，在确定响应函数里，关闭编辑界面，让主界面显示 然后widget_set_text 设置文本显示， 每次文本都能显示出来，但是显示出来后 界面就卡死了 ，尤其是字多的时候几乎每次都卡死，字少不会。
如果是只是在主界面的一个按钮里响应函数中去widget_set_text，不管多少字都能显示且不卡死， 所以应该是我界面切换导致文本显示后就卡死。

static ret_t on_btn_textedit_ok(void* ctx, event_t* evt)
{
	widget_t* win = WIDGET(ctx);
	(void)evt;
        const wchar_t* text = widget_get_text(win);
        window_close(win->parent);
	widget_set_prop_bool(page.par_main, WIDGET_PROP_VISIBLE, TRUE); //
       
      //如果在此处加个延时，那么切换到主界面文本显示后卡死的几率会小，所以想我上面的程序是不是要等到主界面完全显示（或者说重绘完），才能调用widget_set_text去显示文本
       //在此处widget_set_text，让文本显示
｝

xianjimli, 2019-07-02:
建议不要修改awtk本身的代码，否者同步最新的代码很麻烦。你可以用widget_set_prop_xxx在widget上增加新的属性。

ytchw101, 2019-07-02:
好的 明白  只是头文件增加几个成员   为了方便使用    每次更新源码 也会更新自己定义的成员上去

closed 2019-07-05 by ytchw101