zlgopen/awtk #162: 几个问题

ytchw101, 2019-07-01:
1.多行编辑器（mledit） 开启可自动折行, 编辑文字输入空格后再输入文字， 文字超过编辑器宽度后，会换行 在编辑文字超过宽度时下面的字都不显示。
2.多行编辑器能不能增加个垂直滚动条之类，如果编辑文字超过编辑器高度下面文字都看不到了。
3.widget_set_text  如果有空格的话空格不显示， 建议能不能让空格也能显示出来呢。

xianjimli, 2019-07-01:
你用新的代码试试。

widget\_set\_text的问题，我没重现，你试试这个demo:

https://github.com/zlgopen/awtk-c-demos/blob/master/demos/mledit.c

https://github.com/zlgopen/awtk-c-demos/blob/master/demos/edit.c

xianjimli, 2019-07-01:
明天有空的话，我加上垂直滚动条

ytchw101, 2019-07-02:
> 你用新的代码试试。
> 
> widget_set_text的问题，我没重现，你试试这个demo:
> 
> https://github.com/zlgopen/awtk-c-demos/blob/master/demos/mledit.c
> 
> https://github.com/zlgopen/awtk-c-demos/blob/master/demos/edit.c
哦 必须加好几个空格才能看出来效果  比如打一个汉字 加一个空格再打一个汉字 那么显示出来 两个字之间空格基本看不出来有  呵呵

xianjimli, 2019-07-02:
我这样设置也是正常的：
```c
 widget_set_text(edit1, L"中文 　  比 如 打 一 个 汉 字  加 一 个 空 格 再 打 一 个 汉 字   a b c 123");
```

xianjimli, 2019-07-02:
增加了垂直滚动条：https://github.com/zlgopen/awtk/blob/master/demos/assets/raw/ui/mledit.xml

closed 2019-08-12 by ytchw101