zlgopen/awtk #167: demo_ui_app.c 编译 error: no matching function for call to 'widget_on'

oskycar, 2019-07-04:
### clang编译awtk/demo中代码时报了很多类型转换的编译错误
- 1号错误
`demos/demo_ui_app.c:252:15: error: cannot initialize a variable of type 'const char *' with an lvalue of type 'void *'
  const char* name = ctx;
              ^      ~~~`
- 2号错误
`demos/demo_ui_app.c:474:7: error: no matching function for call to 'widget_on'
      widget_on(widget, EVT_CLICK, on_change_locale, "zh_CN"); `

`src/base/widget.h:1079:9: note: candidate function not viable: no known conversion from 'const char [6]' to 'void *' for 4th argument
int32_t widget_on(widget_t* widget, uint32_t type, event_func_t on_event, void* ctx);`

2号错误在demo_tr_app.c  demo_ui_app.c中都有

是不是scons用gcc编译的时候加了什么编译选项，让这些隐式类型转换不认为是错误而只是警告呢。

### 执行awtk/demo中编译出来的可执行程序的目录问题
awtk/demo中代码编译出来的可执行程序直接点击运行时会提示`!!!Asset [name=earth type=image] not exist!!!`,把assets文件夹拷贝到程序所在目录下也不行。只能cd到demo目录后，执行`../bin/demo_animator`才能正常运行，是什么原因？

xianjimli, 2019-07-04:
我加了强制类型转换，你再试试，谢谢。

可执行文件是放在哪个目录的？目前是通过相对路径去查找资源文件的，最好将可执行文件统一放到bin目录下。

oskycar, 2019-07-04:
> 我加了强制类型转换，你再试试，谢谢。
> 
> 可执行文件是放在哪个目录的？目前是通过相对路径去查找资源文件的，最好将可执行文件统一放到bin目录下。

可执行程序是在bin目录下的，是用scons编译出来的。直接双击运行会提示下面截图中的问题。**只能是命令行进入到awtk目录下运行 `./bin/demoui`才能正常打开。**

awtk-c-demos中的程序能直接双击正常运行，我看代码区别是tk_init函数传了RES_ROOT宏
`tk_init(320, 480, APP_SIMULATOR, NULL, RES_ROOT);`
**在awtk的代码中按照这个改了后也能正常运行了，但是可能不是你说的用相对路径的效果。**

[image]

xianjimli, 2019-07-04:
awtk-c-demos没有自己的资源，用的是awtk/demos中的资源，所以把RES_ROOT定义成绝对路径的，绝对路径肯定是可以的。

oskycar, 2019-07-04:
> awtk-c-demos没有自己的资源，用的是awtk/demos中的资源，所以把RES_ROOT定义成绝对路径的，绝对路径肯定是可以的。

但还是不明白相对路径时，awtk/bin中的程序为什么不能直接双击打开运行呢。不知你说的相对路径是怎么个找法？
我把assets文件夹试着拷贝到各种路径下都不行。按理assets文件夹只要和可执行程序在一个文件夹里就行了吧。

xianjimli, 2019-07-04:
Windows上双击是可以运行的。你在MacOS上吗？

oskycar, 2019-07-04:
> Windows上双击是可以运行的。你在MacOS上吗？

是的，macos Mojave 10.14.5 系统

xianjimli, 2019-07-04:
好的。可能是fs\_get\_exe在MacOS上实现有点问题，明天我在MacOS上试试。

xianjimli, 2019-07-04:
已修改，你有空试试。

closed 2019-07-05 by oskycar