zlgopen/awtk #168: 关于编辑器的两个问题

ytchw101, 2019-07-04:
1.更新到最新awtk源码(2019/07/03), 编辑器有时候键盘的删除还是删不了文字。
2.
   单行编辑器设置为
	edit_set_input_type(edit, INPUT_UINT);
        edit_set_int_limit(edit, 8, 200, 1);  

   设置范围为[8, 20], 但实际编辑还是可以输入低于8或超过200的数。

xianjimli, 2019-07-04:
加auto_fix属性，失去焦点时会自动更正。

在输入过程中时没法限制。比如，编辑器开始是空的，用户输入1，1 < 8，理论上是不允许输入的，但是用户本来是想输入12的，禁止输入1，TA就没法输入12。

xianjimli, 2019-07-04:
> 有时候键盘的删除还是删不了文字。

能帮我找些规律吗？谢谢

oskycar, 2019-07-04:
> > 有时候键盘的删除还是删不了文字。
> 
> 能帮我找些规律吗？谢谢

demoui程序中的keyboard页面运行后有以下问题：
1、输入法为En时却是中文输入状态，切换成“中”时变为英文输入状态；
2、软键盘上的删除键不能删除内容，需要敲实体键盘上的退格键；
3、在桌面环境下，比如mac系统下，好像不能使用系统安装的输入法，是不支持还是需要编译时开启什么选项。

[image]

xianjimli, 2019-07-04:
> 1、输入法为En时却是中文输入状态，切换成“中”时变为英文输入状态；

这个我没找到合适的图标，自己换一个吧。图标即要表达当前的"状态"，又要体现"切换"含义，否则有人认为它是当前的"状态"，有人认为它是"切换"的按钮。

> 2、软键盘上的删除键不能删除内容，需要敲实体键盘上的退格键；

已修改。你更新一下。

>3、在桌面环境下，比如mac系统下，好像不能使用系统安装的输入法，是不支持还是需要编译时开启什么选项。

使用APP_DESKTOP标识。

```
tk_init(800, 480, APP_DESKTOP, "AWTK Demo", res_root);
```

请参考demos/demo_desktop.c

ytchw101, 2019-07-04:
> > 有时候键盘的删除还是删不了文字。
> 
> 能帮我找些规律吗？谢谢

仔细试了下是这样：
1.多行编辑器是空的，开始输入文字 ，这样删除没问题。
2.
我觉得是不不是跟我设置mledit焦点有关系   
我程序是切换到编辑界面时,设置了edit_set_focus(edit, TRUE);  这样输入键盘能自动打开，并且widget_set_text 设置edit上有文字。 这个时候点击删除就不行了(偶尔有时可以删)，  不能删的时候你把输入键盘关掉 再打开就又可以删除了

ytchw101, 2019-07-04:
> 加auto_fix属性，失去焦点时会自动更正。
> 
> 在输入过程中时没法限制。比如，编辑器开始是空的，用户输入1，1 < 8，理论上是不允许输入的，但是用户本来是想输入12的，禁止输入1，TA就没法输入12。

哦   我回程序限定下吧  如果awtk那边能实现判断输入内容值，低于就设为最小值等 那更好了  
总之，幸苦了！

xianjimli, 2019-07-04:
删除的问题是我这边的问题，你用最新代码试试。

ytchw101, 2019-07-04:
> 删除的问题是我这边的问题，你用最新代码试试。

用最新的可以了   谢谢！

closed 2019-08-13 by ytchw101