zlgopen/awtk #171: 更新代码后windows上编译报错 STBTT_DEF

oskycar, 2019-07-06:
看error好像是跟STBTT_DEF定义相关的。
`
nanovg.c
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(466) : error C2054: 在“inl
ine”之后应输入“(”
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(470) : error C2085: “stbtt
_BakeFontBitmap”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(480) : error C2085: “stbtt
_aligned_quad”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(482) : error C2082: 形参“i
nline”的重定义
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(482) : error C2143: 语法错
误 : 缺少“;”(在“类型”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(485) : error C2143: 语法错
误 : 缺少“)”(在“*”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(485) : error C2081: “stbtt
_aligned_quad”: 形参表中的名称非法
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(485) : error C2085: “stbtt
_GetBakedQuad”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(485) : error C2143: 语法错
误 : 缺少“;”(在“*”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(486) : error C2059: 语法错
误:“)”
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(519) : error C2054: 在“inl
ine”之后应输入“(”
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(519) : error C2085: “stbtt
_PackBegin”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(530) : error C2082: 形参“i
nline”的重定义
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(530) : error C2143: 语法错
误 : 缺少“;”(在“类型”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(530) : error C2085: “stbtt
_PackEnd”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(535) : error C2082: 形参“i
nline”的重定义
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(535) : error C2143: 语法错
误 : 缺少“;”(在“类型”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(536) : error C2085: “stbtt
_PackFontRange”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(558) : error C2085: “stbtt
_pack_range”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(560) : error C2082: 形参“i
nline”的重定义
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(560) : error C2143: 语法错
误 : 缺少“;”(在“类型”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(560) : error C2143: 语法错
误 : 缺少“)”(在“*”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(560) : error C2081: “stbtt
_pack_range”: 形参表中的名称非法
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(560) : error C2085: “stbtt
_PackFontRanges”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(560) : error C2143: 语法错
误 : 缺少“;”(在“*”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(560) : error C2059: 语法错
误:“)”
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(566) : error C2054: 在“inl
ine”之后应输入“(”
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(566) : error C2085: “stbtt
_PackSetOversampling”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(582) : error C2082: 形参“i
nline”的重定义
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(582) : error C2143: 语法错
误 : 缺少“;”(在“类型”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(585) : error C2143: 语法错
误 : 缺少“)”(在“*”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(585) : error C2081: “stbtt
_aligned_quad”: 形参表中的名称非法
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(585) : error C2085: “stbtt
_GetPackedQuad”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(585) : error C2143: 语法错
[... 118 lines cut ...]
误 : 缺少“;”(在“类型”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(734) : error C2085: “stbtt
_IsGlyphEmpty”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(737) : error C2082: 形参“i
nline”的重定义
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(737) : error C2143: 语法错
误 : 缺少“;”(在“类型”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(737) : error C2143: 语法错
误 : 缺少“)”(在“*”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(737) : error C2081: “stbtt
_vertex”: 形参表中的名称非法
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(737) : error C2085: “stbtt
_GetCodepointShape”: 不在形参表中
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(737) : error C2143: 语法错
误 : 缺少“;”(在“*”的前面)
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(737) : error C2059: 语法错
误:“)”
d:\work\projects\awtk\3rd\nanovg\base\stb_truetype.h(737) : fatal error C1003:
错误计数超过 100；正在停止编译
scons: *** [3rd\nanovg\base\nanovg.obj] Error 2
scons: building terminated because of errors.

D:\work\Projects\awtk>

`

oskycar, 2019-07-06:
visual studio 2013 版本   编译器版本  18.00.21005.1
是不是visualstudio太老了，需要升级

xianjimli, 2019-07-06:
我用的vs2017没问题。

xianjimli, 2019-07-06:
2013 好像不支持inline关键字。

xianjimli, 2019-07-06:
我在linux/macos/windows vs2017/armcc都测试过，这些编译器都支持inline关键字，要不你升级一下吧

oskycar, 2019-07-06:
> 我在linux/macos/windows vs2017/armcc都测试过，这些编译器都支持inline关键字，要不你升级一下吧

我安装了下visualstudio2019 的确可以了。麻烦了。

closed 2019-07-06 by oskycar