zlgopen/awtk #172: 程序运行时能否不显示终端窗口

oskycar, 2019-07-11:
在运行程序的时候总是会打开一个命令行窗体输出一些调试信息，有办法能不显示吗？

xianjimli, 2019-07-11:
定义NDEBUG或者从demos/demo_main.c中删除下面的代码：

```c
#if defined(WIN32)
#if !defined(NDEBUG)
  {
    AllocConsole();
    FILE* fp = NULL;
    freopen_s(&fp, "CONOUT$", "w+t", stdout);
  }
#endif /*NDEBUG*/
#endif /*WIN32*/
```

oskycar, 2019-07-11:
> 定义NDEBUG或者从demos/demo_main.c中删除下面的代码：
> 
> ```c
> #if defined(WIN32)
> #if !defined(NDEBUG)
>   {
>     AllocConsole();
>     FILE* fp = NULL;
>     freopen_s(&fp, "CONOUT$", "w+t", stdout);
>   }
> #endif /*NDEBUG*/
> #endif /*WIN32*/
> ```

感谢

oskycar, 2019-07-11:
> 定义NDEBUG或者从demos/demo_main.c中删除下面的代码：
> 
> ```c
> #if defined(WIN32)
> #if !defined(NDEBUG)
>   {
>     AllocConsole();
>     FILE* fp = NULL;
>     freopen_s(&fp, "CONOUT$", "w+t", stdout);
>   }
> #endif /*NDEBUG*/
> #endif /*WIN32*/
> ```

删了你说的那段代码，但是scons编译出来的程序运行时还是带着命令行窗口。

xianjimli, 2019-07-12:
macos吗？我还不知道怎么搞，先研究一下。