zlgopen/awtk #173: awtk使用请教

ytchw101, 2019-07-11:
2.当前界面关闭后，然后让另外一个界面显示，那么如何判断显示界面某些控件已经完全显示
  出来了。

ytchw101, 2019-07-11:
或者有没什么方法在一个界面关闭前将数据发送给另外一个要显示的界面，并且要在界面完全显示后才能处理数据？

xianjimli, 2019-07-11:
是返回前一个窗口，还是打开新窗口？

ytchw101, 2019-07-12:
2个动态创建的窗口(window_create创建的)，  一个作为主窗口（不可关闭）,  主窗口一个按钮点击创建新窗口（主窗口隐藏） 新窗口显示。新窗口点击返回，显示主窗口关闭新窗口。

ytchw101, 2019-07-12:
我现在遇到问题是：  在关闭新窗口的按钮响应函数里或者在新窗口EVT_WINDOW_CLOSE响应函数里 给 主界面的控件widget_set_text 设置内容（内容较多才会出现比如30多个汉字等），有时会在主界面显示后（settext的内容也显示了）会导致我整个系统挂掉，不知道是我系统哪地方出问题还是？（平台是嵌入式ARM + RTT）

做个另外一个测试：就是只是主界面搞一个按钮，按一次widget_set_text一次内容（内容还比较大月100个汉字）都不会挂掉， 正常。
所以导致我挂掉那就是界面切换时widget_set_text导致的，所以我想是不是要等到我主界面的控件完全显示出来才能widget_set_text？？否则会导致崩溃。

xianjimli, 2019-07-12:
理论上没问题。你搞个最简单的demo，在PC上测试一下，如果还是有问题，发给我看看。

ytchw101, 2019-07-12:
上位机我简单修改了个demo  测试还没发现有问题呢  切换都可以....

ytchw101, 2019-07-17:
现在发现一个诡异情况： 动态建立两个界面  主界面创建了一个二维码(qr_create),  子界面创建个mledit 且edit_set_focus(edit, TRUE); 设为焦点这样键盘自动弹出。   1.我关闭子界面显示主界面,每次主界面显示后我整个程序就挂了。 2. 如果我提前点击键盘右上角的关闭按钮,先关闭键盘再 关闭子界面显示主界面,基本都不会挂，没问题。  3.如果我主界面没有创建二维码，怎么弄程序也不会挂,没问题。  
(平台：嵌入式ARM+RTT+LCD屏 ) 不知道是屏显存原因？ 还是其他什么原因。。。  
或者说有没有如何判断键盘已经完全关闭的方法，或者程序如何动态关闭弹出来的键盘方法？

xianjimli, 2019-07-17:
方便的话，在github上建个仓库，搞一个在PC上可以重现的例子，否则我没法查。

ytchw101, 2019-07-18:
PC例子没搞好,   在demo1_app.c例子里加mledit 
 	widget_t* edit = mledit_create(win, 10, 5, 200, 100); 
        edit_set_focus(edit, TRUE); 
编译运行后一点几mledit就崩了。

我现在想请教下：我工程里 键盘出来后如何动态关闭这个键盘（而不是edit_set_focus），或者说如何获得键盘对象指针/?

xianjimli, 2019-07-18:
试试：input_method_request(input_method(), NULL);

closed 2019-08-12 by ytchw101