zlgopen/awtk #177: widget_clone会导致滚动条异常

poweel, 2019-07-14:
在tab_button_group，scroll_view等需要滚动的控件里面，克隆一个子控件会导致不能正常滚动

下面是我测试的代码

Index: demos/assets/raw/ui/tab_scrollable.xml
===================================================================
--- demos/assets/raw/ui/tab_scrollable.xml	(revision 2335)
+++ demos/assets/raw/ui/tab_scrollable.xml	(working copy)
@@ -1,12 +1,12 @@
 <window anim_hint="htranslate" theme="tab_top_compact">
   <tab_button_group style="scrollable" x="c" y="12" w="90%" h="30" compact="true" scrollable="true">
-    <tab_button text="General"/>
-    <tab_button text="Network" value="true" />
-    <tab_button text="Security"/>
-    <tab_button text="Security1"/>
-    <tab_button text="Security2"/>
-    <tab_button text="Security3"/>
-    <tab_button text="Security4"/>
+    <tab_button name="tab_button_grp1" text="General"/>
+    <tab_button name="tab_button_grp2" text="Network" value="true" />
+    <tab_button name="tab_button_grp3" text="Security"/>
+    <tab_button name="tab_button_grp4" text="Security1"/>
+    <tab_button name="tab_button_grp5" text="Security2"/>
+    <tab_button name="tab_button_grp6" text="Security3"/>
+    <tab_button name="tab_button_grp7" text="Security4"/>
   </tab_button_group>
   <pages x="c" y="42" w="90%" h="-60" value="1">
     <view w="100%" h="100%">
Index: demos/demo_ui_app.c
===================================================================
--- demos/demo_ui_app.c	(revision 2335)
+++ demos/demo_ui_app.c	(working copy)
@@ -90,6 +90,13 @@
 
   install_click_hander(win);
 
+  if (tk_str_eq(name, "tab_scrollable")) {
+    widget_t* tab_button = widget_lookup(win, "tab_button_grp7", TRUE);
+    widget_t* new_tab_button = widget_clone(tab_button, tab_button->parent);
+    widget_set_name(new_tab_button, "tab_button_grp8");
+    tab_button_set_value(tab_button, TRUE);
+  }
+
   if (tk_str_eq(widget_get_type(win), WIDGET_TYPE_DIALOG)) {
     int32_t ret = dialog_modal(win);

xianjimli, 2019-07-15:
需要同时clone button和view，我更新了demoui，你参考一下：

```c
  widget_t* widget = WIDGET(ctx);
  widget_t* button = widget_lookup(widget, "clone_button", TRUE);
  widget_t* view = widget_lookup(widget, "clone_view", TRUE);

  widget_t* new_button = widget_clone(button, button->parent);
  widget_t* new_view = widget_clone(view, view->parent);

```

poweel, 2019-07-16:
我同时clone button和view也不能更新，我发现这个问题的关键是，我的clone动作和tab_button_set_value是在刚打开窗口的时候。
我做了另外一个试验，在打开窗口的时候启动一个idle_ququ，在这个idle_ququ里面启动第二个idle_queue，在第二个idle_queue里面执行tab_button_set_value，就可以了。在第一个idle_queue里面执行都还不生效。
我在打开窗口时，设置一个点击或者键盘事件，在这些事件里面执行 tab_button_set_value 也都正常。

看起来是在刚打开窗口，立刻clone控件，设置tab_button_set_value 会有问题。

poweel, 2019-07-16:
我刚才又试了原始的demo，在多次clone后，也不能更新。

xianjimli, 2019-07-16:
你更新代码，再试试。

closed 2019-07-27 by poweel