zlgopen/awtk #180: 没有做嵌入式用的文件系统适配吗

wcyingdream, 2019-07-16:
我打开raw文件夹里，fs_os.c这个文件里面几乎是空的，在zlgopen这个github上也没找到有关嵌入式文件系统的相关适配，李老师可以做一个吗，比如适配fatfs啥的

xianjimli, 2019-07-16:
这个需要自己包装一下。

wcyingdream, 2019-07-17:
这是我移植fatfs的代码，没成功，希望移植成功的帮我看一下，非常感谢！

`#include "tkc/fs.h"

#ifdef WITH_FS_RES
#include "tkc/mem.h"
#include "tkc/utils.h"
#include "ff.h"
#include "usart.h"

static int32_t fs_os_file_read(fs_file_t* file, void* buffer, uint32_t size) {
  UINT br;
	FIL* fp = (FIL*)(file->data);
	FRESULT res = f_read(fp, buffer, size, &br);
  return (int32_t)res;
}

int32_t fs_os_file_write(fs_file_t* file, const void* buffer, uint32_t size) {
  UINT bw;
	FIL* fp = (FIL*)(file->data);
	FRESULT res = f_write(fp, buffer, size, &bw);
  return (int32_t)res;
}

ret_t fs_os_file_seek(fs_file_t* file, int32_t offset) {
  FIL* fp = (FIL*)(file->data);
	f_lseek(fp, offset);
  return RET_OK ;
}

ret_t fs_os_file_truncate(fs_file_t* file, int32_t size) {
  FIL* fp = (FIL*)(file->data);
	f_sync(fp);           /*If not syncronized fclose can write the truncated part*/
  f_truncate(fp);
  return RET_OK;
}

ret_t fs_os_file_close(fs_file_t* file) {
  FIL* fp = (FIL*)(file->data);
  f_close(fp);
  TKMEM_FREE(file);

  return RET_OK;
}

ret_t fs_os_dir_rewind(fs_dir_t* dir) {
  DIR* d = (DIR*)(dir->data);

  f_rewinddir(d);

  return RET_OK;
}

ret_t fs_os_dir_read(fs_dir_t* dir, fs_item_t* item) {
	FRESULT res;
	FILINFO fno;
  DIR* d = (DIR*)(dir->data);

  memset(item, 0x00, sizeof(fs_item_t));
	
  do {
    	res = f_readdir(d, &fno);
    	if(res != FR_OK) return RET_FAIL;

			if(fno.fattrib & AM_DIR) {
				item->is_dir = 1;
				item->is_file = 0;
				item->name[0] = '/';
				strcpy(&item->name[1], fno.fname);
			}
[... 229 lines cut ...]
static const fs_t s_os_fs = {.open_file = fs_os_open_file,
                             .remove_file = fs_os_remove_file,
                             .file_exist = fs_os_file_exist,
                             .file_rename = fs_os_file_rename,

                             .open_dir = fs_os_open_dir,
                             .remove_dir = fs_os_remove_dir,
                             .dir_exist = fs_os_dir_exist,
                             .dir_rename = fs_os_dir_rename,

                             .get_file_size = fs_os_get_file_size,
                             .get_disk_info = fs_os_get_disk_info,
                             .get_cwd = fs_os_get_cwd,
                             .get_exe = fs_os_get_exe};

fs_t* os_fs(void) {
  return (fs_t*)&s_os_fs;
}
#else
fs_t* os_fs(void) {
  return NULL;
}
#endif

`

xianjimli, 2019-07-17:
你可以创建一个awtk-fatfs的仓库，其他人也可能需要，大家一起来完善。