zlgopen/awtk #181: assets_init函数失败，导致死机

Jun1172, 2019-07-17:
您好：
我现在将awtk相关源文件添加到了我的工程，准备跑一个程序试一下效果。但是到了assets_init函数就会死机，debug追踪到是assets_init里面的tk_init_assets函数，执行switch的时候出了问题。

这个测试的程序，我是先在pc上运行过的，打印一个字符串，pc端可以正常打印出来。
`ret_t application_init() {
	widget_t* win = window_create(NULL, 0, 0, 0, 0);

  char height[] = "40";

  /* 创建文本框*/
  widget_t* label_4_edit = label_create(win, 0, 0, 0, 0);
  widget_set_text(label_4_edit, L"123");
  widget_set_name(label_4_edit, "label_4_edit");
  widget_set_self_layout_params(label_4_edit, "center", "20%", "60%", height);

  widget_layout(win);
}`

请问这里出问题，可能是我哪一步移植有问题？

现在我的移植做了以下的一些事情。
1.添加了awtk的源文件，编译不报错

2.platform里面的get_time_ms，放在一个中断里面，仿真可以监控到它在运行
`uint32_t get_time_ms() {
	awtk_time++;
  return awtk_time;
}`

3.sleep_ms里面提供了一个1ms的延时函数
`void sleep_ms(uint32_t ms) {
  H_Delay(ms);
}`

4.给GUI的内存分配，抄的awtk的一个文档里面的
`#define TK_MEM_SIZE 8 * 1024 * 1024

ret_t platform_prepare(void) {
  uint32_t* mem = (uint32_t*) malloc(TK_MEM_SIZE);
  return_value_if_fail(mem != NULL, RET_OOM);

  tk_mem_init(mem, TK_MEM_SIZE);

  return RET_OK;
}`

5.platform_disaptch_input函数，里面获取了我触摸的数据，其他的抄的awtk里面的内容
`uint8_t platform_disaptch_input(main_loop_t* loop) {

	static int ts_state=TS_ACT_NONE;
	t_msg msg;
	POINT *pt;
	
	TouchGUIMasg_Time();//1.3ms以内
	
	msg.msg64 = TouchProcReadMsg();
	pt->x = (int)(msg.u64.time>>16);//480-
	pt->y = ((int)(msg.u64.time&0x0000ffff));//800-
	int Pressed = msg.u64.val;
	
	/* 通过GTP_Execu获取触摸坐标和状态 */
	if(Pressed > 0)
	{
			ts_state =TS_ACT_DOWN;
	}
	else
	{
		if(ts_state==TS_ACT_DOWN)
		{
			ts_state =TS_ACT_UP;
		}
		else
		{
			ts_state =TS_ACT_NONE;
		}
	}
	if(ts_state != TS_ACT_NONE)
		main_loop_post_pointer_event(loop, TRUE, pt->x, pt->y);
	else
		main_loop_post_pointer_event(loop, FALSE, pt->x, pt->y);

  return 0;
}`

6.platform_create_lcd里面调用了lcd_mem_rgba8888_create_double_fb
`lcd_t* platform_create_lcd(wh_t w, wh_t h) {
  return lcd_mem_rgba8888_create_double_fb(w, h, online_fb_addr, offline_fb_addr);
}`

Jun1172, 2019-07-17:
查明原因，堆栈溢出

soyzhc, 2019-08-05:
awtk 的 GUI 线程建议分配 >=32KB 的栈

closed 2019-08-08 by xianjimli