zlgopen/awtk #184: 动画刷新导致奔溃问题

linany, 2019-07-19:
ret_t widget_animator_manager_time_elapse(widget_animator_manager_t* am, uint32_t delta_time) {
  widget_animator_t* iter = NULL;
  return_value_if_fail(am != NULL, RET_BAD_PARAMS);

  delta_time = delta_time * am->time_scale;

  iter = am->first;
  while (iter != NULL) {
    widget_animator_t* next = iter->next;   //这里

    widget_animator_time_elapse(iter, delta_time);
    iter = next;
  }

  return RET_OK;
}
这个next拿出来，有可能在widget_animator_time_elapse被destroy了，会有问题，slide_view里一些例子是这么用的，有时候会崩溃。

xianjimli, 2019-07-19:
没看出来有问题啊。widget_animator_time_elapse只会destroy它自己(iter)，不会destroy next。所以我提前把next保存下来了，next应该是有效的。

linany, 2019-07-19:
demoui中的slide_view_v1，左边那个slide_indicator 如果有auto_hide，在slide_view，滑动动画结束时会destroy一个自动dis visible的动画。这个动画就是next的东西
if (slide_indicator->wa_opactiy != NULL) {
    widget_animator_destroy(slide_indicator->wa_opactiy);
    slide_indicator->wa_opactiy = NULL;
  }

closed 2019-10-15 by linany