zlgopen/awtk #188: 二维码+问题咨询

ytchw101, 2019-07-26:
1.widget_t* edit = mledit_create(...)创建一个mledit, 控件

ytchw101, 2019-07-26:
> 1.widget_t* edit = mledit_create(...)创建一个mledit, 控件 未编辑任何文字
const wchar_t *ptext = widget_get_text(edit);
int len = wcslen(ptext); //为什么得到的len不是0
widget_set_text(edit, L""); 之后 得到的len 才会是0

问题2:创建一个二维码控件
void test(void)
｛
  assets_manager_t* rm = assets_manager();
  assets_manager_load(rm, ASSET_TYPE_FONT, "default");
  awtk_set_locale_simplified_chinese();
  widget_t *new_qr_code = qr_create(cont_obj, 20, 20, 80, 80);
  qr_set_pixsize(new_qr_code, 3);
  char tr_str[64];
  int value = 123456;
  const char* format = locale_info_tr(locale_info(), "中文%d");
  qr_set_text(new_qr_code, tr_str);
  widget_set_name(new_qr_code, "new_qr_code");
  widget_use_style(new_qr_code, "qr_border");
  widget_set_enable(new_qr_code, TRUE);
｝

问题：二维码是显示出来的但是我扫描后为什么中文是乱码,是哪地方没搞好吗？

问题3：我想获得二维码的点数据 我直接qr_t* qr = QR(new_qr_code); 的话得到的qr->qr_code_data是空,qr->qr_width 是 0所以我 如下操作 ：
	qr_t* qr = QR(new_qr_code);
	if (qr->qr_text != NULL) 
	{
			QRcode *qr_code;
			qr_code = QRcode_encodeString((const char *)qr->qr_text, 3, QR_ECLEVEL_H, 
                                                                                                                           QR_MODE_8, 1);
			qr->qr_code_data = qr_code;
			qr->qr_width = qr_code->width;
           }
可以得到了宽度和数据。
那么1.能像上面那样操作来获得数据吗？
       2.这个qr->qr_code_data  需要我手动QRcode_free吗？感觉我用时有时会导致我系统崩溃

ytchw101, 2019-07-26:
打印出
	for(i = 0; i < strlen(tr_str); i++)
	{
			rt_kprintf("%x ,", tr_str[i]);
	}
输出:d6 ,d0 ,ce ,c4 ,31 ,32 ,33 ,34 ,35 ,36    扫码中文还是乱的

xianjimli, 2019-07-27:
第一个问题已修改，请用最新代码验证。

第二个问题请发到二维码项目上。

ytchw101, 2019-08-01:
发到二维码项目 在哪发？

xianjimli, 2019-08-01:
https://github.com/CUITzhaoqi/awtk-qr

closed 2019-08-12 by ytchw101