zlgopen/awtk #193: hscroll_label里面的字体设置不了颜色

poweel, 2019-08-02:
hscroll_label里面的字体设置不了颜色，同样的ui描述文件，把标签换成label就可以了，hscroll_label设置颜色是有什么特殊的操作吗？

xianjimli, 2019-08-02:
我试了，可以啊：
```
<hscroll_label>
  <style name="default" text_color="black">
    <normal   />
    <focused  />
  </style>

  <style name="green" text_color="green">
    <normal   />
    <focused  />
  </style>
</hscroll_label>
```

poweel, 2019-08-03:
用style里面的样式可以，用内嵌样式不行。

xianjimli, 2019-08-04:
你怎么用的

poweel, 2019-08-04:
<hscroll_label x="left" y="0" w="80" h="34"  name="_line_str_4" only_focus="true" text="12345678" text_color="#00ff00"/>
我又试了label标签也不行
设置style是正常的，我就设置style吧

closed 2019-08-04 by poweel

xianjimli, 2019-08-04:
inline style 请参考：https://github.com/zlgopen/awtk/blob/master/docs/theme.md

```
style:normal:font_size="16"
```