zlgopen/awtk #196: mledit EVT_VALUE_CHANGING事件为什么没反应

ytchw101, 2019-08-08:
如题 注册了 EVT_VALUE_CHANGING事件，	widget_on(edit, EVT_VALUE_CHANGING, on_edit_changing, edit);  响应函数里没响应。

xianjimli, 2019-08-08:
支持了EVT_VALUE_CHANGING事件，你更新代码试试。

ytchw101, 2019-08-09:
可以了 但是还有个问题就是  为什么windget_set_text(edit, text); 不能触发EVT_VALUE_CHANGING事件，

xianjimli, 2019-08-09:
好的，我加上。

xianjimli, 2019-08-10:
已加。你更新代码试试。

ytchw101, 2019-08-14:
更新代码试了， windget_set_text后没有触发EVT_VALUE_CHANGED ？？？

另外还有个问题就是： 按键删除字符不触发EVT_VALUE_CHANGING事件？

xianjimli, 2019-08-14:
windget_set_text有变化才触发。

xianjimli, 2019-08-14:
> 另外还有个问题就是： 按键删除字符不触发EVT_VALUE_CHANGING事件？

你更新代码，再试试。

ytchw101, 2019-08-14:
更新代码  按键删除字符可以触发EVT_VALUE_CHANGING事件了
但是
1. 从一个界面返回到有mledit的界面 ，然后windget_set_text  （那就意味着 mledit的内容从空到有）但是并没有触发EVT_VALUE_CHANGED事件， 而是等到我关闭这个mledit的界面才触发？
2.当触发了EVT_VALUE_CHANGING事件后，就一直处于EVT_VALUE_CHANGING触发状态（即使我我没有编辑或删除文字），会一直循环进入事件响应函数！？

closed 2019-08-15 by ytchw101