zlgopen/awtk #199: dialog.c有的接口为什么删掉

ytchw101, 2019-08-08:
如题 更新最新源码（19/08/08），提示diaog.c中有的接口没有实现？

最主要 更新源码后，编译通过，烧写进去 屏始终不显示，但程序应该没死。 不知为什么？

每次更新源码 我都是替换之前的 src和3rd文件夹, 其他文件还是按以前的保留，

ytchw101, 2019-08-08:
目前的工程awtk源码是7/20号左右的， 之前更新源码 我都是这样操作，现在操作了2 、3遍  控件还是显示不了，   不清楚是没移植好还是说我这边还需要什么配置

感觉像是修改这个之后我这个屏显示不了了
（
window manager抽象成接口。
引入native_window接口。）

xianjimli, 2019-08-08:
最新版本需要增加以下文件：
src/window_manager/window_manager_default.c
src/native_window/native_window_raw.c
src/widgets/dialog_helper.c

ytchw101, 2019-08-12:
不知道为什么更新最新源码（19/08/10版本）我的lcd屏还是不显示（花屏）， 
新添加以下文件
src/window_manager/window_manager_default.c
src/native_window/native_window_raw.c
src/widgets/dialog_helper.c

ytchw101, 2019-08-12:
还有个src/base/native_window.c   添加以上文件后顺利通过编译，我的其他代码未动,  更新源码我一般是直接替换 src和3rd文件夹 ，   但是烧到卡上 lcd始终花屏， 版本用没改之前的显示都可以  ！  目前正在找原因。。。。。不知道更新后我的其他哪方面要作何更改。。。

closed 2019-08-13 by ytchw101