zlgopen/awtk #209: 关于EVT_VALUE_CHANAGING再请教

ytchw101, 2019-08-15:
更新代码 按键删除字符可以触发EVT_VALUE_CHANGING事件了
但是

1.从一个界面返回到有mledit的界面 ，然后windget_set_text （那就意味着 mledit的内容从空到有）但是并没有触发EVT_VALUE_CHANGED事件， 而是等到我关闭这个mledit的界面才触发？
2.当触发了EVT_VALUE_CHANGING事件后，就一直处于EVT_VALUE_CHANGING触发状态（即使我我没有编辑或删除文字），会一直循环进入事件响应函数！？

xianjimli, 2019-08-16:
没有重现。你自己调试一下，或者把可以运行的代码贴上来。

ytchw101, 2019-08-19:
例子：
	widget_on(edit, EVT_VALUE_CHANGING, on_edit_changing, edit);  
	widget_on(edit, EVT_VALUE_CHANGED, on_edit_changed, edit);  

static ret_t on_edit_changed(void* ctx, event_t* e) 
{
  widget_t* win = (widget_t*)ctx;
  (void)e;

  rt_kprintf("on_edit_changed\r\n");
  return RET_OK;
｝

从另外一个界面切换到此界面并
windget_set_text(edit, L"1234");后未触发on_edit_changed  当我点击关闭此界面时，打印
on_edit_changed

2.第二个问题是我搞错了， 我是同一个界面创建一个mledit 和view控件(view_create创建)，并注册
widget_on(edit, EVT_VALUE_CHANGING, on_edit_changing, edit); 
widget_on(view, EVT_PAINT, on_paint_view, edit); //test   当mledit内容改变时，在on_paint_view中绘制些东西。  发现当触发EVT_VALUE_CHANGING后  view就一直重绘。

static ret_t on_paint_view(void* ctx, event_t* e) 
{
  widget_t* win = (widget_t*)ctx;
  (void)e;
  
  rt_kprintf("on_paint_view\r\n");
  return RET_OK;
｝

xianjimli, 2019-08-21:
我测试没问题啊。你参考一下我写的代码：https://github.com/zlgopen/awtk-c-demos/blob/master/demos/mledit_set_text.c

closed 2019-08-29 by ytchw101