zlgopen/awtk #212: Ubuntu14.4下编译报错了，且相关依赖安装不上去

yongzhena, 2019-08-16:
3rd/SDL/gen/pointer-constraints-unstable-v1-protocol.c:29:26: fatal error: wayland-util.h: No such file or directory
 #include "wayland-util.h"

AWTK依赖了很多第三方的库吗？
知道AWTK很强大，但是能否整理出一份精简版的呢？比如针对stm32的话可能很多代码和特性是不需要的，但是也都柔和在一块儿。
结果新人可能卡在环境和编译上。移植也是无从下手。
如果有一份最小的精简内容最好，
让能在其他环境下容易编译通过。

yongzhena, 2019-08-16:
yang@yang-vir:/mnt/awtk/awtk$ sudo apt-get install  libsndio-dev libgtk-3-dev libglu1-mesa libglu1-mesa-dev libgl1-mesa-glx libgl1-mesa-dev 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libroar-dev' instead of 'libsndio-dev'
libglu1-mesa is already the newest version.
libglu1-mesa set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 empathy : Depends: libclutter-1.0-0 (>= 1.10.0) but it is not going to be installed
           Depends: libclutter-gst-2.0-0 (>= 0.10.0) but it is not going to be installed
           Depends: libclutter-gtk-1.0-0 (>= 1.4) but it is not going to be installed
           Depends: libcogl15 (>= 1.15.8) but it is not going to be installed
           Recommends: gvfs-backends
 libcheese-gtk23 : Depends: libclutter-1.0-0 (>= 1.13.2) but it is not going to be installed
                   Depends: libclutter-gtk-1.0-0 (>= 0.91.8) but it is not going to be installed
                   Depends: libcogl15 (>= 1.15.8) but it is not going to be installed
 libcheese7 : Depends: libclutter-gst-2.0-0 (>= 0.10.0) but it is not going to be installed
              Depends: gstreamer1.0-clutter but it is not going to be installed
 libegl1-mesa : Conflicts: libegl1-x11
 libegl1-mesa-lts-xenial : Depends: libgl1-mesa-dri-lts-xenial (= 11.2.0-1ubuntu2~trusty1) but it is not going to be installed
                           Conflicts: libegl1-mesa
                           Conflicts: libegl1-x11
 libgl1-mesa-dev : Depends: mesa-common-dev (= 10.1.0-4ubuntu5) but it is not going to be installed
                   Depends: libdrm-dev (>= 2.4.52) but it is not going to be installed
                   Depends: libxdamage-dev but it is not going to be installed
                   Depends: libxext-dev but it is not going to be installed
                   Depends: libxfixes-dev but it is not going to be installed
 libgl1-mesa-glx : Depends: libglapi-mesa (= 10.1.0-4ubuntu5)
 libgtk-3-dev : Depends: libgtk-3-0 (= 3.10.8-0ubuntu1) but 3.10.8-0ubuntu1.6 is to be installed
                Depends: gir1.2-gtk-3.0 (= 3.10.8-0ubuntu1) but 3.10.8-0ubuntu1.6 is to be installed
                Depends: libglib2.0-dev (>= 2.37.5) but it is not going to be installed
                Depends: libgdk-pixbuf2.0-dev (>= 2.27.1) but it is not going to be installed
                Depends: libpango1.0-dev (>= 1.32.4) but it is not going to be installed
                Depends: libatk1.0-dev (>= 2.7.5) but it is not going to be installed
                Depends: libatk-bridge2.0-dev but it is not going to be installed
                Depends: libcairo2-dev (>= 1.13.0~20140204) but it is not going to be installed
                Depends: libxext-dev but it is not going to be installed
                Depends: libxinerama-dev but it is not going to be installed
                Depends: libxi-dev but it is not going to be installed
                Depends: libxrandr-dev but it is not going to be installed
                Depends: libxcursor-dev but it is not going to be installed
                Depends: libxfixes-dev but it is not going to be installed
                Depends: libxcomposite-dev but it is not going to be installed
                Depends: libxdamage-dev but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

xianjimli, 2019-08-16:
需要 Ubuntu需要 16 以上。 14太旧了，安装19吧。

soyzhc, 2019-08-17:
stm32 有专门移植好的工程
https://github.com/zlgopen/awtk-stm32f429igtx-raw
https://github.com/zlgopen/awtk-stm32f103ze-raw