zlgopen/awtk #213: float_t在不同平台上长度不一致，导致bsvg文件与平台相关

soyzhc, 2019-08-19:
见 types_def.h
```
#if defined(WITH_DOUBLE_FLOAT)
typedef long double float_t;
#else
typedef float float_t;
#endif /*WITH_DOUBLE_FLOAT*/
```
通常情况下 float_t 是4字节
在定义 WITH_DOUBLE_FLOAT 的情况下，float_t 是12或16字节，见 awtk_config.py
```
elif OS_NAME == 'Linux':
  if TARGET_ARCH == 'x86':
    OS_FLAGS = OS_FLAGS + ' -DWITH_DOUBLE_FLOAT '

  elif TOOLS_NAME == 'mingw' :
    COMMON_CCFLAGS=COMMON_CCFLAGS+' -DWITH_DOUBLE_FLOAT -DUNICODE ' 
```
bsvg 文件里面直接用到 float_t  作为结构体成员，会导致 bsvg 文件是平台相关的
如果我在上述平台调试好app，把然后把工程搬到 arm 平台使用，会出现异常

建议把 bsvg 改为平台无关的结构，或者把 float_t 改为所有平台长度固定的类型

xianjimli, 2019-08-19:
好的，已修改。谢谢

soyzhc, 2019-08-20:
还有个问题就是
```
elif OS_NAME == 'Linux':
  if TARGET_ARCH == 'x86':
    OS_FLAGS = OS_FLAGS + ' -DWITH_DOUBLE_FLOAT '
```
在 Linux x86 的情况下，不应该加 -DWITH_DOUBLE_FLOAT
如果加了就会导致 float_t 是 8 字节，与编译器内置的 4 字节冲突（已经在树莓派 Linux x86 上试验）

xianjimli, 2019-08-20:
可能与版本有关，当时加上这个就是因为32bit 的系统中，把float_t定义成了double。我想想怎么搞。

soyzhc, 2019-08-21:
float_t 受到 FLT_EVAL_METHOD 影响
FLT_EVAL_METHOD=0 时, float_t 是4字节
FLT_EVAL_METHOD=2 时, float_t 是12字节

很奇怪的是, 在 x86 Linux 中 FLT_EVAL_METHOD=2
但在其他 Linux(包括 ARM 32bit Linux) 或 mac 中 FLT_EVAL_METHOD=0

这是网上的一段解释
```
在 x87 结构的浮点数运算器下，FLT_EVAL_METHOD 默认为 2；在 K&R C 中，FLT_EVAL_METHOD 的默认值为 1；在其它大多数情况下，FLT_EVAL_METHOD 的默认值为 0。
对于 GCC，在 32 位 x86 环境下，FLT_EVAL_METHOD 默认为 2；在 64 位 x86 环境下，FLT_EVAL_METHOD 默认为 0（可以通过编译选项 -mfpmath=387 将 FLT_EVAL_METHOD 的值设置为 2）。
```

这个问题可能无解, 只能根据当前的编译平台, 选择是否启用 -DWITH_DOUBLE_FLOAT

soyzhc, 2019-08-21:
不过有一点是可以确定的: float_t 语义上就是可变长的, 
所以不应该在交换文件的结构定义中使用 float_t
否则会导致这种文件与平台相关

closed 2019-08-29 by soyzhc