zlgopen/awtk #214: 怎么给 Label 设置下划线  ?

V1ki, 2019-08-21:
Label 有办法 设置下划线  , 如果不行的话,有没有其他比较便捷的方法来设置下划线.

V1ki, 2019-08-21:
找到方法了,可以将border 设置bottom 以设置下划线.

V1ki, 2019-08-21:
虽然可以使用border 来设置下划线,但是只有border 在不停的闪烁,请问 border的实现原理是什么?

xianjimli, 2019-08-22:
什么平台？我在Linux上测试没问题。

V1ki, 2019-08-22:
> 什么平台？我在Linux上测试没问题。

海思的HiFB

xianjimli, 2019-08-22:
border没有什么特殊的啊。

V1ki, 2019-08-22:
> border没有什么特殊的啊。

感觉有可能是因为border的高度太低了,有办法设置border的高度吗?

xianjimli, 2019-08-22:
你修改一下这行代码试试：

widget.c:1083         canvas_draw_hline(c, 0, h - 1, w);

V1ki, 2019-08-22:
> canvas_draw_hline(c, 0, h - 1, w);

canvas_draw_hline 这个方法好像并不能直接设置hline 的高度,我把这个地方该成fill_rect 试试.

xianjimli, 2019-08-22:
你的高度指线宽还是y的坐标？

V1ki, 2019-08-22:
> 你的高度指线宽还是y的坐标？

vgcanvas_set_line_width(canvas, 1);  
按照这个来说,应该是指线宽

V1ki, 2019-08-22:
将之前绘制底部border的方法改成如下方法,并且在style 中设置 ``border_width``  为大一些的值,就不会出现非常严重的闪烁.

```
      if (border & BORDER_BOTTOM)
      {
        canvas_set_fill_color(c, bd);
        canvas_fill_rect(c, 0, h-border_width, w, border_width);
        // canvas_draw_hline(c, 0, h - 1, w);
      }
```

closed 2019-08-22 by V1ki