zlgopen/awtk #215: src\app_main.obj : fatal error LNK1112: 模块计算机类型“x64”与目标计算机类型“X86”冲突

zzzh, 2019-08-22:
在awtk-examples\HelloWorld-Demo里面执行scons时，会出现这个错误。
奇怪的是，在awtk里面运行scons是正常的

xianjimli, 2019-08-22:
什么平台？试试awtk-hello是否正常

zzzh, 2019-08-22:
windows 7 平台， 好的，我下载awtk-hello试试

zzzh, 2019-08-22:
试过了，也是同样错误

xianjimli, 2019-08-22:
32位系统吗？

zzzh, 2019-08-22:
64位

xianjimli, 2019-08-22:
我们测试了几台机器没有重现。麻烦把scons完整的log贴上来看看。

soyzhc, 2019-08-29:
确认一下python是否也是安装64位版本

zzzh, 2019-09-14:
@xianjimli 
scons: Reading SConscript files ...
ARCH=('32bit', 'WindowsPE') TARGET_ARCH=x86
TK_ROOT: G:\pdf\play\lab\awtk
scons: done reading SConscript files.
scons: Building targets ...
link /MACHINE:X86 /DEBUG /OUT:bin\hello_awtk.exe /LIBPATH:lib /LIBPATH:G:\pdf\pl
ay\lab\awtk\lib assets.lib awtk.lib gpinyin.lib awtk.lib linebreak.lib nanovg-ag
ge.lib nanovg.lib agge.lib SDL2.lib glad.lib gdi32.lib user32.lib winmm.lib imm3
2.lib version.lib shell32.lib ole32.lib Oleaut32.lib Advapi32.lib DelayImp.lib p
sapi.lib src\assets.obj src\awtk_main.obj src\main.obj src\window1.obj
Microsoft (R) Incremental Linker Version 14.00.24215.1
Copyright (C) Microsoft Corporation.  All rights reserved.

src\assets.obj : fatal error LNK1112: 模块计算机类型“x64”与目标计算机类型“X86
”冲突
scons: *** [bin\hello_awtk.exe] Error 1112
scons: building terminated because of errors.

zzzh, 2019-09-14:
@soyzhc 果然python是32位，换成64位可以编译了，但最后出错如下：
scons: Reading SConscript files ...
ARCH=('64bit', 'WindowsPE') TARGET_ARCH=
TK_ROOT: G:\pdf\play\lab\awtk
scons: done reading SConscript files.
scons: Building targets ...
link /MACHINE:X64 /DEBUG /OUT:bin\hello_awtk.exe /LIBPATH:lib /LIBPATH:G:\pdf\pl
ay\lab\awtk\lib assets.lib awtk.lib gpinyin.lib awtk.lib linebreak.lib nanovg-ag
ge.lib nanovg.lib agge.lib SDL2.lib glad.lib gdi32.lib user32.lib winmm.lib imm3
2.lib version.lib shell32.lib ole32.lib Oleaut32.lib Advapi32.lib DelayImp.lib p
sapi.lib src\assets.obj src\awtk_main.obj src\main.obj src\window1.obj
Microsoft (R) Incremental Linker Version 14.00.24215.1
Copyright (C) Microsoft Corporation.  All rights reserved.

assets.obj : error LNK2019: 无法解析的外部符号 assets_manager，该符号在函数 asse
ts_init 中被引用
assets.obj : error LNK2019: 无法解析的外部符号 assets_manager_preload，该符号在
函数 assets_init 中被引用
assets.obj : error LNK2019: 无法解析的外部符号 tk_init_assets，该符号在函数 asse
ts_init 中被引用
awtk_main.obj : error LNK2019: 无法解析的外部符号 path_app_root，该符号在函数 Wi
nMain 中被引用
awtk_main.obj : error LNK2019: 无法解析的外部符号 tk_init，该符号在函数 WinMain
中被引用
awtk_main.obj : error LNK2019: 无法解析的外部符号 tk_run，该符号在函数 WinMain
中被引用
awtk_main.obj : error LNK2019: 无法解析的外部符号 tk_ext_widgets_init，该符号在
函数 WinMain 中被引用
main.obj : error LNK2019: 无法解析的外部符号 widget_child_on，该符号在函数 appli
cation_init 中被引用
window1.obj : error LNK2001: 无法解析的外部符号 widget_child_on
main.obj : error LNK2019: 无法解析的外部符号 window_open，该符号在函数 applicati
on_init 中被引用
window1.obj : error LNK2001: 无法解析的外部符号 window_open
window1.obj : error LNK2019: 无法解析的外部符号 window_close，该符号在函数 on_cl
ose_window 中被引用
G:\pdf\play\lab\awtk\lib\assets.lib : warning LNK4272:库计算机类型“X86”与目标
计算机类型“x64”冲突
G:\pdf\play\lab\awtk\lib\awtk.lib : warning LNK4272:库计算机类型“X86”与目标计
算机类型“x64”冲突
G:\pdf\play\lab\awtk\lib\gpinyin.lib : warning LNK4272:库计算机类型“X86”与目标
计算机类型“x64”冲突
G:\pdf\play\lab\awtk\lib\linebreak.lib : warning LNK4272:库计算机类型“X86”与目
标计算机类型“x64”冲突
G:\pdf\play\lab\awtk\lib\nanovg-agge.lib : warning LNK4272:库计算机类型“X86”与
目标计算机类型“x64”冲突
G:\pdf\play\lab\awtk\lib\nanovg.lib : warning LNK4272:库计算机类型“X86”与目标
计算机类型“x64”冲突
G:\pdf\play\lab\awtk\lib\agge.lib : warning LNK4272:库计算机类型“X86”与目标计
算机类型“x64”冲突
G:\pdf\play\lab\awtk\lib\SDL2.lib : warning LNK4272:库计算机类型“X86”与目标计
算机类型“x64”冲突
G:\pdf\play\lab\awtk\lib\glad.lib : warning LNK4272:库计算机类型“X86”与目标计
算机类型“x64”冲突
bin\hello_awtk.exe : fatal error LNK1120: 10 个无法解析的外部命令
scons: *** [bin\hello_awtk.exe] Error 1120
scons: building terminated because of errors.

xianjimli, 2019-09-14:
清除以前编译的，再重新编译试试：
```
scons -c
scons 
```