zlgopen/awtk #217: 请问如何最佳实现类似Android GridView的效果？

webnoise, 2019-08-25:
请问如何最佳实现类似Android GridView的效果？用list_view勉强能做，但是很不方便。

xianjimli, 2019-08-26:
有千个item不是问题，最多1M内存。上千个不同的图片，可能是个大问题。

xianjimli, 2019-08-26:
目前还不支持虚拟滚动，有同事在研究。

webnoise, 2019-08-26:
请问如何最佳实现类似Android GridView的效果（多个item的layout相同，同一行可包含多个Item）？用list_view勉强能做，但是很不方便。我看了一下demo都是写死的list_item内部嵌入多个children的方式来实现的。

xianjimli, 2019-08-27:
可以看看AWTK-MVVM，用MVVM只需要写一条模板，其它的根据数据自动生成。

webnoise, 2019-08-30:
可能是我得问题表达得不清楚，我也看了mvvm中的books.xml相关的demo。但是该demo依然跟我要表达的问题有出入,如books.xml依然一行只能显示一个list_item，我想让一行中含有两个甚至多个相同的listitem（Android中的GridView），但感觉atwk没有这方面的控件支持，不知道为什么不实现这么常用的控件。

webnoise, 2019-08-30:
[image]

xianjimli, 2019-08-31:
children_layout_list_view增加了一个cols参数，你看看是不是你要的：
```
<window anim_hint="htranslate">
  <list_view x="0"  y="0" w="100%" h="-50"> 
    <scroll_view name="view" x="0"  y="0" w="-12" h="100%"
      children_layout="list_view(item_height=128,cols=3,spacing=2)">
```

https://github.com/zlgopen/awtk/blob/master/demos/assets/raw/ui/image_list.xml

webnoise, 2019-09-02:
前两天太忙，今天试了一下，效果很棒！！！

closed 2019-09-02 by webnoise

YinWenshuo, 2020-04-08:
请问下，你上面那个缩略图是怎么做的？每个list_item都贴一张图？