zlgopen/awtk #218: label 设置 style 并invalidate  后, 并没达成style 效果.

V1ki, 2019-08-26:
首先,我想做成就是收到 按键事件的时候,将 焦点 (也就是设置背景) 集中到选中的地方上去,

但是目前遇到一个问题,在 海思HiFB 平台下, 有些 label 无法正常的设置背景 (动态的 ,直接在xml 中设置style的时候,并没有任何问题),  目前比较疑惑,不知道是在哪方面出的问题. 

界面的控制代码如下:

```

char *label_names[9] = {"exit_label", "screen_sel_1cam", "screen_sel_2cam", "screen_sel_comp", "osd_switch_on", "osd_switch_off", "sd_format_switch_no", "sd_format_switch_yes", "set_label"};

int current_focus_index = 0;

void focus_option(int option_index, bool_t is_select)
{
    widget_t *wm = window_manager();

    widget_t *label = widget_lookup(wm, label_names[option_index], TRUE);
    printf("set label_name:%s style = %s\n", label_names[option_index], is_select ? "menu_item_selected_bg" : "menu_item");

    widget_use_style(label, is_select ? "menu_item_selected_bg" : "menu_item");
    widget_invalidate(label, NULL);

}

void focus_up()
{

    focus_option(current_focus_index, FALSE);
    current_focus_index--;
    if (current_focus_index < 0)
    {
        current_focus_index = 8;
    }
    focus_option(current_focus_index, TRUE);
}

void focus_down()
{

    focus_option(current_focus_index, FALSE);
    current_focus_index++;
    if (current_focus_index >= 9)
    {
        current_focus_index = 0;
    }
    focus_option(current_focus_index, TRUE);
}
```

xml 中的代码如下:

```
<window anim_hint="htranslate">
    <label name="menu_title" x="260" y="30" w="200" h="60"  text="MAINTENANCE  MENU" style="menu_title"/>
    <label name="exit_label" x="100" y="120" w="180" h="50" text="EXIT" style="menu_item_selected_bg"/>
    <label name="screen_sel" x="100" y="200" w="180" h="50" text="SCREEN SEL    " style="menu_item"/>
    <label x="280" y="200" w="30" h="50" text=" : " style="menu_item"/>

    <label name="screen_sel_1cam" x="310" y="200" w="80" h="50" text ="1CAM" style="menu_item"/>
    
    <label x="390" y="200" w="20" h="50" text="/ " style="menu_item"/>

    <label name="screen_sel_2cam" x="410" y="200" w="80" h="50" text ="2CAM"  style="menu_item_selected"/>

    <label x="490" y="200" w="20" h="50" text="/ " style="menu_item"/>

    <label name="screen_sel_comp" x="510" y="200" w="100" h="50" text ="COMP"  style="menu_item"/>

    <label name="osd_switch" x="100" y="280" w="180" h="50" text="OSD ON / OFF  "  style="menu_item"/>
    <label x="280" y="280" w="30" h="50" text=" : " style="menu_item"/>

    <label name="osd_switch_on" x="310" y="280" w="70" h="50" text ="ON" style="menu_item_selected"/>
    
    <label x="380" y="280" w="20" h="50" text="/ " style="menu_item"/>

    <label name="osd_switch_off" x="400" y="280" w="70" h="50" text ="OFF"  style="menu_item"/>

    <label name="sd_format_switch" x="100" y="350" w="180"  text="SD FORMAT" h="50"  style="menu_item"/>
    <label x="280" y="350" w="30" h="50" text=" : " style="menu_item"/>

    
    <label name="sd_format_switch_no" x="310" y="350" w="70" h="50" text ="NO" style="menu_item_selected"/>
    
    <label x="380" y="350" w="20" h="50" text="/ " style="menu_item"/>

    <label name="sd_format_switch_yes" x="400" y="350" w="70" h="50" text ="YES"  style="menu_item"/>

    <label name="set_label" x="100" y="420" w="180" h="50" text="SET" style="menu_item"/>

</window>

```

tangMans, 2019-08-27:
第一，在你的XML文件中发现有三个“menu_item_selected_bg”，"menu_item"，"menu_item_selected"样式，不知道你特意这样写的还是误写了？

第二，你提到“有些 label 无法正常的设置背景”，也不知道是不是因为第一点引起的？

第三，建议你不要弄这么多个label先，可以弄两三个label试试用widget_use_style设置样式是否可行

第四，在void focus_option(int option_index, bool_t is_select)函数中你固定通过设置false或true，来设置样式觉得有点问题吧，如果你想达到当某个焦点在某个label上就显示一种特别的样式，而其他的显示另外一种样式的话，可以显示设置，不用搞到这么复杂吧

V1ki, 2019-08-27:
> 第一，在你的XML文件中发现有三个“menu_item_selected_bg”，"menu_item"，"menu_item_selected"样式，不知道你特意这样写的还是误写了？
>
> 第二，你提到“有些 label 无法正常的设置背景”，也不知道是不是因为第一点引起的？
>
> 第三，建议你不要弄这么多个label先，可以弄两三个label试试用widget_use_style设置样式是否可行 [...]

1.  关于第一点.这个是故意这样写的.
2.  关于第二点,个人感觉并不是 由于第一点影响的 ,
3.  关于第三点, 目前情况下 `exit_label` 以及 `set_label` 肯定会生效,但是有时生效也会出错.
4. 关于第四点  `menu_item_selected_bg` 下 有背景(焦点). 当接收到上下左右的命令时,会将焦点进行移动, 同时需要将其他的焦点给取消掉.
5 . 关于第四点. 目前我并不太清楚这种设置 样式是否会有问题,但是由于界面需要,无法使用list_view . 但是 我在list_view上使用如上的设置样式的方法,可以达成  我想要的焦点的效果.

xianjimli, 2019-08-27:
遇到问题，搞个最简测试环境去验证怀疑的点，才容易定位问题。

如果搞不定，在github上建一个可以编译执行的工程，发给我试试。

V1ki, 2019-09-04:
在定位问题发现,当出现焦点,或者 style 设置不生效的时候,由于没有过多时间纠结这个问题,所以采用 canvas 来解决此问题.  等后续有时间后,再来分析具体问题.

closed 2019-09-04 by V1ki