zlgopen/awtk #219: 请教按钮显示图片问题？

ytchw101, 2019-08-26:
1.我在一个界面创建了8个按钮，每个按钮风格中指定不同的图片，开机显示时有种较为明显感觉是图片一个个出来， 有没有什么更好的方法能实现图片像是同时出来的？

2.另外还有其他按钮widget_set_text  如果是设置中文也是和1一样的感觉 (比刷图片快点)，英文要快一点， 不知道有什么更好的方法没？

平台：嵌入式ARM

xianjimli, 2019-08-26:
像demoui一样，预先加载图片，避免一般显示一边加载。

xianjimli, 2019-08-26:
什么硬件配置，加载字体都这么慢呢？

ytchw101, 2019-08-27:
哦 但预加载图片和之前没预加载感觉差不多 不知到我有没弄错？？？     步骤如下：

1.在assets_init中添加图片资资源 如下：
        assets_manager_add(rm, image_text);
	assets_manager_add(rm, image_count);
	assets_manager_add(rm, image_datetime);
	assets_manager_add(rm, image_barcode);
	assets_manager_add(rm, image_bmp);
	assets_manager_add(rm, image_print);
	assets_manager_add(rm, image_split);
2.程序开始时预加载图片，如下：
	       bitmap_t img[6];
	       ret_t res;
               image_manager_get_bitmap(image_manager(), "text", &img[0]); 
		image_manager_get_bitmap(image_manager(), "count", &img[1]);
		image_manager_get_bitmap(image_manager(), "datetime", &img[2]);
		image_manager_get_bitmap(image_manager(), "barcode", &img[3]);
		image_manager_get_bitmap(image_manager(), "bmp", &img[4]);
		res = image_manager_get_bitmap(image_manager(), "print", &img[5]);
		rt_kprintf("image_manager_get_bitmap res end=%d\r\n", res);

3.打开主界面并查找到对应图片按钮，且设置文本
        page.par_main = window_open("page_main"); 
	 /**********************************图片按钮*******************************************/
	  widget_t* btn_text = widget_lookup(page.par_main, "btn_text", TRUE); 
	  widget_t* btn_count = widget_lookup(page.par_main, "btn_count", TRUE); 
	  widget_t* btn_time = widget_lookup(page.par_main, "btn_time", TRUE); 
	  widget_t* btn_barcode = widget_lookup(page.par_main, "btn_barcode", TRUE); 
	  widget_t* btn_bmp = widget_lookup(page.par_main, "btn_bmp", TRUE); 
	  widget_t* btn_print = widget_lookup(page.par_main, "btn_print", TRUE); 

	  widget_set_text(label_text, L"文本");
	  widget_set_text(label_count, L"计数");
	  widget_set_text(label_time, L"日期");
	  widget_set_text(label_barcode, L"条码");
	  widget_set_text(label_bmp, L"图形");
	  widget_set_text(label_print, L"打印");	

注：在styles的 default中 指定了每个按钮对应的icon

显示整体感觉效果像是从左往右一个个出来，  没有想象中的刷的一下所有图片同步贴上去的感觉，

平台：嵌入式ARM（900MHz + 512kRAM + 8M flash）

xianjimli, 2019-08-27:
LCD分辨率多大，如何与MCU连接的(i2c?)。

ytchw101, 2019-08-27:
480*272    问了硬件 LCD是40pin 接到ARM 上    应该不是读写数据速度问题...

另外我上面的步骤去预加载 再创建 显示 有什么纰漏吗？？？

xianjimli, 2019-08-27:
参考这个文件，看看各个显示函数花费的时间： https://github.com/zlgopen/awtk/blob/master/docs/optimation.md

closed 2019-09-02 by ytchw101