zlgopen/awtk #220: 如何使用3个framebuffer

lzy2259, 2019-08-28:
看了文档里的framebuffer介绍可以开启3个framebuffer提高fps。请问要如何使用3个framebuffer呢？我看linux fb的移植里面只有双framebuffer的代码。

xianjimli, 2019-08-28:
可以参考：https://github.com/zlgopen/awtk-stm32f429igtx-raw/blob/master/USER/main.c

lzy2259, 2019-08-28:
参考了一下列子。我的板子只有一个fb，所以只把原来的lcd_mem_bgra8888_create_duble_fb换成lcd_mem_bgra8888_create_three_fb。我搜索了一下全部的代码为什么next_fb都没有看到在哪个地方调用。

xianjimli, 2019-08-28:
```
ret_t lcd_stmf429_swap(lcd_t* lcd) {
  lcd_mem_t* mem = (lcd_mem_t*)lcd;

  s_next_fb = mem->offline_fb;

  return RET_OK;
}

void HAL_LTDC_LineEvenCallback(LTDC_HandleTypeDef* LTDC_Handler) {
  if (s_next_fb != NULL) {
    HAL_LTDC_SetAddress(LTDC_Handler, (uint32_t)s_next_fb, 0);
    __HAL_LTDC_RELOAD_CONFIG(LTDC_Handler);
    s_online_fb = s_next_fb;
    s_next_fb = NULL;
  }

  HAL_LTDC_ProgramLineEvent(LTDC_Handler, 0);
}
```

xianjimli, 2019-08-28:
中断里处理的，linux得另外开一个线程，比较麻烦。