zlgopen/awtk #222: 关于widget_destroy_children使用问题

ytchw101, 2019-08-29:
在程序中使用widget_destroy_children可能导致程序崩溃，

问题叙述：
在主界面创建了个  cont_obj = canvas_widget_create(page.main, 0, 23, 480, 150);
在cont_obj创建一个或多个其他控件，  当从另外一个界面返回到主界面之前先
widget_destroy_children(cont_obj); 删除cont_obj上的所有对象  。

程序崩溃时，我系统提示(data abort 即数据终止)，所以一开始怀疑自己哪里数据处理有问题，断断续续找了1个多星期原因，没找到后来去掉widget_destroy_children， 改为一个个删除（widget_destroy） 目前测试多次正常， 所以怀疑是widget_destroy_children导致崩溃。

xianjimli, 2019-08-29:
widget_destroy_children很简单，而且是一个频繁调用的函数，出错的概率不大，我在valgrind上也跑过，没有发现问题。你做一个可以重现的最小demo，我帮你查一下。

ytchw101, 2019-08-29:
改成widget_destroy  测试目前没有出现过崩溃 ，  
可能前面说的不够详细，   
1.主界面创建个cont_obj作为容器，在cont_obj上创建label控件， 字体大小190；
2.从另外一个界面切换到主界面前，调用widget_destroy_children删除cont_obj下的所有子项；
3.再在cont_obj上创建新的label控件并设置文本(字体大小190)
4,有时点击（在我lcd屏上我是用手划下label控件）会导致崩溃，有时试了23十次都不出现，有时一点就出现，具有偶然性。

ytchw101, 2019-08-29:
另外现在改成widget_destroy  发现个疑问：代码如下

widget_t* page_main  = NULL;
widget_t* cont_obj = NULL;
static ret_t on_btn_test(void* ctx, event_t* evt)
{	
     int32_t count = widget_count_children(cont_obj);
	
		 log_debug("count=%d\r\n", count);
		 int32_t i;
		 widget_t* pchild = NULL;
		 for(i = 0; i < count; i++)
		 {
				pchild = widget_get_child(cont_obj, i);
			 
				log_debug("i=%d, pchild=%ld\r\n", i, cont_obj);
			 
				if(pchild)
				{
					 widget_destroy(pchild);
					 pchild = NULL;
				}  
		 }
		 
//     widget_destroy_children(cont_obj);
		 count = widget_count_children(cont_obj);
		 log_debug("count2=%d\r\n", count);
		 	
	return RET_OK;
}

void test(void)
{
    page_main = window_create(NULL, 0, 0, 0, 0);
		
		cont_obj = canvas_widget_create(page_main, 5, 5, 200, 100);
	  widget_set_style_str(cont_obj, "bg_color", "black"); //±³¾°É«
		
		widget_t* label1 = label_create(cont_obj, 10, 10, 80, 30);
		widget_set_text(label1, L"ÎÄ±¾1");
		
		widget_t* label2 = label_create(cont_obj, 100, 10, 80, 30);
		widget_set_text(label2, L"ÎÄ±¾2");
		
		widget_set_style_str(label1, "text_color", "white");
		widget_set_style_str(label2, "text_color", "white");
		
		
		widget_t* btn_test = button_create(page_main, 80, 120, 80,30);
		widget_set_text(btn_test, L"²âÊÔ");
		
		widget_on(btn_test, EVT_CLICK, on_btn_test, NULL);
}

在上位机上运行结果：

count=2
i=0, pchild=57443296
widget_get_child:2336 index < widget->children->size
i=1, pchild=57443296
count2=1
为什么获得第二个子项和第一个地址一样？widget_get_child索引不是从0开始吗？？， 所以第二个不能删除掉。

ytchw101, 2019-08-29:
widget_destroy_children 问题我回头写个小demo
 但   我的平台是 ARM+LCD ， 且加载了.ttf字体， 上面创建的label对象我都是设置字体大小190左右, 小字体的话 好像不容易重现...

ytchw101, 2019-08-29:
哦 wiget_destroy 删除一项后，索引就变化了  。。。

closed 2019-09-10 by ytchw101