zlgopen/awtk #223: 为什么创建mledit控件会引起其他控件一直重绘

ytchw101, 2019-08-29:
代码：
static ret_t on_view_paint(void* ctx, event_t* e) {
	rt_kprintf("on_view_paint...\r\n");
	return RET_OK;
}
static ret_t on_cont_paint(void* ctx, event_t* e) 
{      
      rt_kprintf("on_cont_paint...\r\n");
	return RET_OK;
}
void test(void)
{
          widget_t* par_main = window_create(NULL, 0, 0, 0, 0);
	  widget_set_style_str(par_main, "bg_color", "black"); 
	
	  widget_t* edit = mledit_create(par_main, 10, 60, 230, 50); 
	  mledit_set_focus(edit, TRUE); //
	
          widget_t*  view = view_create(par_main, 2, 2, 80, 40);
	  widget_set_style_str(view, "bg_color", "white");

	  widget_t*  cont_obj = canvas_widget_create(par_main, 90, 2, 80, 40);
	  widget_set_style_str(cont_obj, "bg_color", "blue");

	  widget_on(cont_obj, EVT_PAINT, on_cont_paint, NULL); 
          widget_on(view, EVT_PAINT, on_view_paint, NULL); 
//    widget_on(edit, EVT_VALUE_CHANGING, on_edit_changing, edit);
}

运行后会一直打印:
on_view_paint...
on_cont_paint...
on_view_paint...
on_cont_paint...
.......
不创建mledit 则不会， 感觉是光标闪烁引起重绘， 。。。。！！

xianjimli, 2019-08-29:
有system bar吗？

xianjimli, 2019-08-29:
应该是没启用脏矩形，在PC上使用AGGE后端(自动启用脏矩形)时，不会出现上述情况：
https://github.com/zlgopen/awtk-c-demos/blob/master/demos/issue_223.c

ytchw101, 2019-08-29:
没有

ytchw101, 2019-08-29:
启用脏矩形！  怎么启用？   不懂脏矩形......

xianjimli, 2019-08-30:
把你创建LCD的代码贴出来看看。

ytchw101, 2019-08-30:
1.    
lcd_t* platform_create_lcd(wh_t w, wh_t h) {
  return stm32f429_create_lcd(w, h);
}
2.

//#define FB_ADDR (uint8_t*)0X43e89000
//#define FB_ADDR (uint8_t*)0X81f80000
#define FB_ADDR (uint8_t*)0X80800000

lcd_t* stm32f429_create_lcd(wh_t w, wh_t h) {
  lcd_t* lcd = NULL;
  uint32_t size = w * h * 4;
  s_framebuffers[0] = FB_ADDR;
  s_framebuffers[1] = FB_ADDR + size;
  s_framebuffers[2] = FB_ADDR + 2 * size;

//#if LCD_PIXFORMAT == LCD_PIXFORMAT_ARGB8888 || LCD_PIXFORMAT == LCD_PIXFORMAT_RGB888
  lcd = lcd_mem_bgra8888_create_three_fb(w, h, s_framebuffers[0], s_framebuffers[1], s_framebuffers[2]);
                                         
//#else
//  lcd = lcd_mem_bgr565_create_three_fb(w, h, s_framebuffers[0], s_framebuffers[1], s_framebuffers[2]);
//	lcd = lcd_mem_bgr565_create_double_fb(w, h, s_framebuffers[0]);
//#endif /*LCD_PIXFORMAT*/

//  lcd =  lcd_mem_bgra8888_create_single_fb(w, h, s_framebuffers[0]);
	
  lcd->swap = lcd_stmf429_swap;
  lcd->begin_frame = lcd_stmf429_begin_frame;

//  HAL_NVIC_SetPriority(LTDC_IRQn, 1, 1);
//  HAL_NVIC_EnableIRQ(LTDC_IRQn);
//  HAL_LTDC_ProgramLineEvent(&LTDC_Handler, 0);
  HAL_LTDC_SetAddress(0, (uint32_t)s_framebuffers[0], 0);
//  __HAL_LTDC_RELOAD_CONFIG(&LTDC_Handler);

  return lcd;
}

ytchw101, 2019-08-30:
其他子函数：

static uint8_t* s_next_fb = NULL;
static uint8_t* s_online_fb = NULL;
static uint8_t* s_framebuffers[3];

ret_t lcd_stmf429_begin_frame(lcd_t* lcd, rect_t* dirty_rect) {
  if (lcd_is_swappable(lcd)) {
    uint32_t i = 0;
    lcd_mem_t* mem = (lcd_mem_t*)lcd;

    mem->next_fb = NULL;		
    mem->online_fb = NULL;
    mem->offline_fb = NULL;
    for (i = 0; i < ARRAY_SIZE(s_framebuffers); i++) {
      uint8_t* iter = s_framebuffers[i];
      if (iter != s_online_fb && iter != s_next_fb) {
        mem->offline_fb = iter;
        break;
      }
    }
  }

  return RET_OK;
}

ret_t lcd_stmf429_swap(lcd_t* lcd) {
  lcd_mem_t* mem = (lcd_mem_t*)lcd;

  s_next_fb = mem->offline_fb;

  return RET_OK;
}

void HAL_LTDC_SetAddress(uint32_t *hltdc, uint32_t Address, uint32_t LayerIdx)
{
	uint32_t tmpaddr;
//	rt_kprintf("--- LTDC_SetAddress 0x%08x ---\r\n", Address);
	tmpaddr = Address<<3;
//	rt_kprintf("--- LTDC_    ddress 0x%08x ---\r\n", tmpaddr );
	*(volatile unsigned int *) 0x01e60850 = (tmpaddr);
}

void HAL_LTDC_LineEvenCallback(void) {
  if (s_next_fb != NULL) {
    HAL_LTDC_SetAddress(0, (uint32_t)s_next_fb, 0);
//    __HAL_LTDC_RELOAD_CONFIG(LTDC_Handler);
    s_online_fb = s_next_fb;
    s_next_fb = NULL;
  }

//  HAL_LTDC_ProgramLineEvent(LTDC_Handler, 0);
}

xianjimli, 2019-08-30:
3fb不会启用脏矩形。使用2fb吧，参考：https://github.com/zlgopen/awtk-stm32f767igtx-raw/blob/master/USER/main.c

ytchw101, 2019-09-02:
搞定,  之前的一直重绘问题没有了， 而且以前反映的开机刷新慢也变得很快  并且拖动控件也不像之前那样卡顿   看来我这个平台2fb最合适

ytchw101, 2019-09-02:
多谢！

xianjimli, 2019-09-02:
好的，不客气。

closed 2019-09-02 by ytchw101